ros-visualization/rviz #1028: [Mac OS X] Problem Report pops up every time RViz quits

spmaniato, 2016-06-23:
Initially reported at mikepurvis/ros-install-osx#43 

On macOS, an `rviz quit unexpectedly` problem report pops up every single time RViz quits (see screenshot below) It happens both when I quit using the GUI and when I `Ctrl-C` from the terminal. Mac OS X El Capitan `10.11.5` More details appear in the screenshot.

The report's message reads:

> terminating with uncaught exception of type boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >: boost: mutex lock failed in pthread_mutex_lock: Invalid argument
> abort() called

In the terminal, I see

```
libc++abi.dylib: terminating with uncaught exception of type boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >: boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Abort trap: 6
```

Might be somewhat related to #782

[image]

wjwwood, 2016-10-18:
If you run into this again, please copy all of the text in the crash report and post it in a gist.github.com.

spmaniato, 2016-10-18:
@wjwwood That's easy since it happens every time I kill RViz: https://gist.github.com/spmaniato/e4d4c4c9b7da33270269080cabb4adc5 

Thanks for looking into it 😁

wjwwood, 2016-10-18:
Looks like it is crashing in `console_bridge`: https://gist.github.com/spmaniato/e4d4c4c9b7da33270269080cabb4adc5#file-problem-details-and-system-configuration-L43

I don't have time to reproduce it right now, but can you just try making sure you have the latest version of console bridge installed? I think it is installed from Homebrew.

spmaniato, 2016-10-18:
```
$ brew info console_bridge
ros/deps/console_bridge: stable 0.2.5
http://wiki.ros.org/console_bridge
/usr/local/Cellar/console_bridge/0.2.5 (7 files, 55.0K) *
  Built from source on 2016-01-28 at 17:40:45
```

@wjwwood Looking at https://github.com/ros/console_bridge/releases, that does look kinda old 🤔 Digging deeper, it looks like the formula hasn't been updated in a while: [ros/homebrew-deps/console_bridge.rb](https://github.com/ros/homebrew-deps/blob/master/console_bridge.rb) Is that something I can help with in a PR or is it up to you?

PS. I don't think this is just me btw. In mikepurvis/ros-install-osx#43, Mike Purvis says:

> I get this all the time. I've never paid it too much mind since it's more of a nuisance than something which impacts use.

Thanks again!

wjwwood, 2016-10-18:
Well, it's not that old if you compare it to what Indigo uses on trusty:

http://packages.ubuntu.com/source/trusty/console-bridge (0.2.5)

Xenial has 0.3.2, but that *shouldn't* make a difference. You could try to upgrade the version in Homebrew and see if that helps, but I'm not sure that's the root cause of the problem or if the problem is in console bridge that it has been fixed since 0.2.5.