ros-visualization/rviz #1030: Segmentation fault (core dumped) , when run rviz

QinZiwen, 2016-07-14:
I use Ubuntu 14.04, ros indigo. When I run rviz rviz , some error shows:
$ rosrun rviz rviz 
[ INFO] [1468497103.225780415]: rviz version 1.11.14
[ INFO] [1468497103.225844553]: compiled against Qt version 4.8.6
[ INFO] [1468497103.225863189]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1468497103.496038712]: Stereo is NOT SUPPORTED
[ INFO] [1468497103.496135132]: OpenGl version: 3 (GLSL 1.3).
Segmentation fault (core dumped)

what should I do? 
Thanks

wjwwood, 2016-07-14:
That's not enough information for me to figure out the issue, please have a look at: http://wiki.ros.org/rviz/Troubleshooting#How_to_make_a_usable_bug_report

QinZiwen, 2016-07-15:
Yes, I following this tutorials. Some message shows :
    $ rosrun rviz system_info.sh
    [rosrun] Couldn't find executable named system_info.sh below /opt/ros/indigo/share/rviz
    $ rosrun rviz rviz -l > Ogre.log
    *** Error in `/opt/ros/indigo/lib/rviz/rviz': corrupted double-linked list: 0x000000000153cca0 ***
    Aborted (core dumped)
    [orge.txt](https://github.com/ros-visualization/rviz/files/365470/orge.txt)

wjwwood, 2016-07-19:
I need the core file mentioned in that post, otherwise I have no idea why it segfaulted. Alternatively you can run rviz with gdb and paste the output of the `backtrace` command after it crashes.

QinZiwen, 2016-07-20:
@wjwwood I don't know how to get the core file. Could you give me the detail steps or commands. Thank you!

wjwwood, 2016-07-25:
Instructions are in the link I sent you.

From http://wiki.ros.org/rviz/Troubleshooting#How_to_make_a_usable_bug_report:
> Core file. A core file is a dump of all the memory in use by a program when it crashed, so gives even more info than a stack trace. To get a core file (in linux, specifically in the "bash" shell) do:
> - ulimit -c unlimited to remove the core file size limit (default is size 0).
> - run rviz and cause the crash.
> - a file named "core" or similar (sometimes with a process ID number in the name) should appear in the current directory. Include the core file.

wjwwood, 2016-10-18:
I'm going to close this until you can provide some more feedback. Just post here and I'll reopen the issue if needed.

closed 2016-10-18 by wjwwood