ros-visualization/rviz #1032: Error while compiling: sip: Unable to find file "QtWidgets/QtWidgetsmod.sip"

jmlago91, 2016-07-18:
Hi!

I'm getting this error while compiling RViz in Ubuntu 14.04 with ROS Indigo:

[ 68%] Running SIP generator for rviz_sip Python bindings...
[100%] Built target rviz_default_plugin
sip: Unable to find file "QtWidgets/QtWidgetsmod.sip"
Traceback (most recent call last):
  File "/opt/ros/indigo/share/python_qt_binding/cmake/sip_configure.py", line 79, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/home/ctag/catkin_ws_rviz/devel/bin/sip/rviz_sip', '-b', '/home/ctag/catkin_ws_rviz/devel/bin/sip/rviz_sip/pyqtscripting.sbf', '-I', '/usr/share/sip/PyQt4', '-w', '-x', 'VendorID', '-t', 'WS_X11', '-x', 'PyQt_NoPrintRangeBug', '-t', 'Qt_4_8_4', '-x', 'Py_v3', '-g', 'rviz.sip']' returned non-zero exit status 1
make[2]: *** [/home/ctag/catkin_ws_rviz/devel/bin/sip/rviz_sip/Makefile] Error 1
make[1]: *** [rviz/src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I have PyQt5 installed, and QtWidgets/QtWidgetsmod.sip is in his directoty,

Any idea?

Thanks for all.

wjwwood, 2016-07-18:
The indigo version of rviz does not support Qt5 with Python. You'll have to use qt4/pyqt4 or the kinetic version of rviz.

closed 2016-07-18 by wjwwood

jmlago91, 2016-07-18:
Thanks for the reply!

Making qt4 the default qt version and using pyqt4 should it compile?
I have also pyqt4 installed and in the default installation directory the is no QtWidgets folder with QtWidgetsmod.sip.

Thanks again.

jmlago91, 2016-07-19:
There is no problem with qt4/pyqt4.
I have now rviz working.

Thanks for all!

rohin28, 2017-04-09:
Hi @jmlago91 !

Can you list the steps about how did you make pyqt4 the default version? I am trying to install rqt from source on my Raspberry Pi 3 and am getting the same error.

Any help would be welcome!

jmlago91, 2017-04-10:
Remove old versions of PyQT, and try this:
http://pyqt.sourceforge.net/Docs/PyQt4/installation.html