ros-visualization/rviz #1038: Incorrect sphere scaling with SPHERE_LIST marker type

miguelprada, 2016-08-03:
According to the [docs](http://wiki.ros.org/rviz/DisplayTypes/Marker#Sphere_List_.28SPHERE_LIST.3D7.29), when using `SPHERE_LIST` marker type, all spheres must share a common scale (and color). However, my recent experience shows me that irregular scaling (i.e. to render ellipsoids) does not work neither when using the `SPHERE_LIST` type.

You can see this behavior by running [this test package](https://github.com/miguelprada/test_sphere_scale).

Am I doing something wrong? Is this intended behaviour? And if it is, shouldn't it be mentioned in the wiki?

I've tested this on Indigo/Ubuntu12.04. Maybe someone can confirm whether this is still present in later releases.

wjwwood, 2016-08-09:
Thanks for the detailed report. Yeah, looks like the sphere list's scale doesn't work. I don't have time to look into this now, but if you want, it should be relatively easy to fix (I guess). The culprit code is probably here:

https://github.com/ros-visualization/rviz/blob/0d423d6249a39f7f3ddd591007952c6dc940b6ac/src/rviz/default_plugin/markers/points_marker.cpp#L83

It's reusing the PointCloud interface in rviz, so you might look in there:

https://github.com/ros-visualization/rviz/blob/0d423d6249a39f7f3ddd591007952c6dc940b6ac/src/rviz/ogre_helpers/point_cloud.cpp#L346

Also, you might check to see if the same problem exists for `CUBE_LIST`.

miguelprada, 2016-08-09:
I only stumbled upon this because `rviz_visual_tools` was using the `SHPERE_LIST` marker type to publish even a single sphere marker. Since davetcoleman/rviz_visual_tools#47 already fixes the problem for a single sphere marker, I don't really have a strong need to fix it nor a lot of spare time to look into it. I will if I can, but I can't promise anything.

andrewjong, 2022-03-30:
Are there any updates to this? It still doesn't work as of 2022.

rhaschke, 2022-03-30:
@andrewjong, you are very welcome to provide a fixing PR 😄

rhaschke, 2022-03-31:
I shortly looked into the issue a little deeper. The problem is more fundamental:
For efficiency, rviz renders a large number of spheres (as in point clouds or the `SPHERE_LIST` marker type) using a fragment shader ([`shaded_circle.frag`](https://github.com/ros-visualization/rviz/blob/697ed3ff0496b79b1bb4270aabf457ee2f2ae736/ogre_media/materials/glsl120/shaded_circle.frag)), visualizing a simple shaded disk oriented towards the camera. Here is the material definition:
https://github.com/ros-visualization/rviz/blob/697ed3ff0496b79b1bb4270aabf457ee2f2ae736/ogre_media/materials/scripts120/point_cloud_sphere.material#L1-L10

To fix this, one could try to replace the current vertex shader ([billboard.vert](https://github.com/ros-visualization/rviz/blob/697ed3ff0496b79b1bb4270aabf457ee2f2ae736/ogre_media/materials/glsl120/nogp/billboard.vert#L25)) to a shader that ignores the camera position (see [box.vert](https://github.com/ros-visualization/rviz/blob/783256d8e16af5d36a95f064e77e80dd3bec78ae/ogre_media/materials/glsl120/nogp/box.vert)) and adapt the texture coordinates such that a correctly tilted ellipse will be rendered instead.

I won't have time to tackle this. Anybody in need of this feature should tackle this.