ros-visualization/rviz #1045: Metadata from Collada files

de-vri-es, 2016-09-08:
Currently, metadata from Collada files such as the '<up_axis>` and `<unit>` tags are ignored. This is done by ignoring the root transformation on imported models [1].

While this avoids the problem of Assimp enforcing Y_UP orientation, it also undoes the effect of the `<unit>` tag. Additionally, there is the question if ignoring the `<up_axis>` tag is correct behaviour in the first place.

This came up at ros-planning/geometric_shapes#52, because currently `geometric_shapes` imports Collada meshes differently from rviz. That causes an undesirable mismatch between visualization and for example the meshes used for collision checking in moveit.

@gavanderhoorn pointed out that the proposed fix there (which copies rviz behaviour) basically ignores parts of the Collada specification (the `<up_axis>` and `<unit>` tags). However, Assimp does not provide the needed information right now to enforce a convention other than `Y_UP`. It would require parsing the Collada file manually to retrieve the relevant metadata tag.

`geometric_shapes` should load models the same way rviz does to prevent mismatches with visualization. However, if the behaviour of rviz is going to change in the near future, it could make sense to delay changing `geometric_shapes` until after that happened.

So, the real question: Should rviz respect `<unit>` and `<up_axis>` tags in Collada files, and reorient meshes to Z_UP convention and scale them to meter units?

[1] https://github.com/ros-visualization/rviz/blob/0d423d6249a39f7f3ddd591007952c6dc940b6ac/src/rviz/mesh_loader.cpp#L232

de-vri-es, 2016-09-08:
Note that if the answer is "yes, it should" I can put it into a PR. I wanted to discuss first though.

wjwwood, 2016-10-18:
I don't feel strongly about it. Ideally it would follow the standard, but I'm not sure it's worth potentially breaking behavior for existing models that have worked around this issue in someway.