ros-visualization/rviz #1047: Rviz doesn't load default config file.

Kaju-Bubanja, 2016-09-15:
[rviz_l_info.txt](https://github.com/ros-visualization/rviz/files/474148/rviz_l_info.txt)
[rviz_system_info.txt](https://github.com/ros-visualization/rviz/files/474149/rviz_system_info.txt)
[default.rviz.txt](https://github.com/ros-visualization/rviz/files/474160/default.rviz.txt)

When I start rviz with rosrun rviz rviz it doesn't load the default config under .rviz/default.rviz I just get 

[image]

I can add panels and rviz shows all the things after I add the panels and everything is working as normal. When I hit save config it even saves the config to the default.rviz file. I see that the default.rviz changes and correctly saves the state, but when I close rviz and reopen it, I get the picture above every time.
When I try to use the open config option just nothing happens. It is really annoying to always have to reconfigure all the panels and other things.
I already tried to prune and reinstall rviz but this didn't help.
I'm using ros-indigo and Ubuntu 14.04. I don't know why the sys_info command doesn't give any output.

Femer, 2016-09-19:
Hi, I have exactly the same problem: when I open rviz it doesn't load any configuration files and I have the same black screen as you have. Moreover, even tough I add some panels I am not able to rotate the camera view.
I am running ROS-Indigo, on a Ubuntu 14.04 installed on a macbook pro (dual boot, no virtual machine).

Kaju-Bubanja, 2016-09-19:
The not being able to move the camera can be solved by clicking the small plus and adding interact with mouse. I don't know why this is an option you have to explicitly, like who doesn't want to interact with the mouse.

Femer, 2016-09-19:
Yeah, you're right about the camera control.
I found this old issue: [Issue 546](https://github.com/ros-visualization/rviz/issues/546) where they suggest to remove the .rviz folder and run rviz again, but this does not help in my case. I can't load a saved configuration file.

Rahtron3030, 2016-09-20:
I'm having the same problem as well with ROS-Indigo, Ubuntu 14.04 on a macbook pro with dual boot. Any work-arounds?

Kaju-Bubanja, 2016-09-21:
Not yet.

zhangzichao, 2016-10-06:
same problem here...
I end it putting the rviz repository in my workspace, switching to the correct branch and compiling it. Then `rosrun rviz rviz` will call the version that is compiled from the source and it works.

marco-tranzatto, 2016-10-07:
@zhangzichao yeah, you're right!
 If I compile it in my catkin workspace it works! Thanks for the hint! :)

wjwwood, 2016-10-17:
It's strange to me that compiling it from source fixes the issue. Can you guys try doing an `sudo apt-get dist-upgrade`? Perhaps it is an ABI breaking issue that doesn't cause a crash but instead incorrect behavior.

I'm going to close this until we can find an actual problem. Please continue to discuss the issue in the meantime.

closed 2016-10-17 by wjwwood

marco-tranzatto, 2016-10-18:
@wjwwood this fixes the problem as well, now I don't need to compile rviz in my workspace anymore!
Thanks :)

ctchervenkov, 2016-11-10:
Also have the same problem and compiling from source doesn't seem to fix it.
Any updates on an explanation/solution?

wjwwood, 2016-11-10:
@ctchervenkov make sure you have all the latest packages, i.e. do a `sudo apt-get update` and then a `sudo apt-get dist-upgrade`.

ctchervenkov, 2016-11-11:
I made sure everything is up to date, but still no luck

tysik, 2017-01-19:
In my case the same problem happened (on two machines) after installing cartographer_ros (using steps provided in: https://google-cartographer-ros.readthedocs.io/en/latest/). Maybe this could provide some clue. Unfortunately `sudo apt-get dist-upgrade` did not work.

ljarin, 2017-04-22:
just wanted to report that I have the same problem. It happened to me on 2 different machines, perhaps after installing openCV 3 from source

zhehangT, 2017-05-13:
I have the same problem. I still try to work it out.

zhangzichao, 2017-05-14:
For my case, it is because I source my own workspace before launching rviz. And in the workspace I have some package that provides a different version of yaml-cpp from the system one. Then when I launch rviz after sourcing the workspace, the loader will first find the yaml-cpp in my workspace but not the system one. In this case, the RVIZ is compiled against one version and running with another version, which causes the problem.

If anyone wonders if it is the similar case for him/her, one quick way is to launch a fresh bash (make sure only ROS is sourced) and launch RVIZ to see if it works.

zhehangT, 2017-05-14:
It is seemed that the yaml-cpp version will cause the problem. I remove the yaml-cpp I install manual and rviz can work properly.

FaiScofield, 2018-01-28:
I had this problem too.
And I found that there are 2 version of yaml-cpp in my computer, one was installed by cmake. and after I deleted it, my rviz repaired.

hariharan382, 2022-10-15:
For God sake, please mention the location where I can find yaml-cpp file please

rhaschke, 2022-10-15:
`yaml-cpp`  is a library installed on Ubuntu via package `libyaml-cpp-dev`. It should be automatically installed as a dependeny of rviz. People here had issues, because they installed _another_ version of that library. So, you should check which library is actually loaded in your case (if you have issues). This is the expected result:
```sh
ldd /opt/ros/noetic/lib/librviz.so | grep yaml-cpp
	libyaml-cpp.so.0.6 => /lib/x86_64-linux-gnu/libyaml-cpp.so.0.6 (0x00007f8b2ced9000)
```

hariharan382, 2022-10-15:
My problem is whenever I load the rviz ,it is opening default.rviz config
file , but I am having my own config file which is not loading

rhaschke, 2022-10-15:
When started as `rviz -d /path/to/some/file.rviz`, rviz will load the specified file.
If started without that option, rviz will load `$HOME/.rviz/default.rviz` (if available) and otherwise a system default.