ros-visualization/rviz #1049: Ogre exception duplicated resource name

phil0stine, 2016-09-26:
This applies to a non-standard configuration (Ogre 1.10 on OSX and rviz branch ogre_1_9), but it seems like it might apply more broadly so I am reporting.

After successful compilation, running rviz gives the error:

`OGRE EXCEPTION(4:ItemIdentityException): Resource with the name rviz/glsl120/include/pass_depth.vert already exists. in ResourceManager::add at /tmp/ogre/OgreMain/src/OgreResourceManager.cpp (line 168)`

Looking at the GLSL .program files, there do appear to be 2 instances of the same name, `rviz/glsl120/include/pass_depth.vert`, in 
[glsl120](https://github.com/ros-visualization/rviz/blob/0d423d6249a39f7f3ddd591007952c6dc940b6ac/ogre_media/materials/glsl120/glsl120.program) and [gsl120/nogp](https://github.com/ros-visualization/rviz/blob/0d423d6249a39f7f3ddd591007952c6dc940b6ac/ogre_media/materials/glsl120/nogp/nogp.program).

My naive workaround was to rename the resource in `glsl120/nogp` to be `rviz/glsl120/nogp/pass_depth.vert`, which resolved the exception and rviz seems to be working fine.

wjwwood, 2016-10-17:
Can you open a pull request to this effect. It sounds like a worthwhile change.

tanay-bits, 2018-04-19:
I'm still seeing this error despite recompiling after applying #1063. My OGRE version is 1.10.11 (Xalafu), and I'm using an older rviz (version 1.11.8 on indigo). The exact error I get is:

> [ERROR] [1524151129.545731714]: ItemIdentityException: Resource with the name rviz/glsl120/include/pass_depth.vert already exists. in ResourceManager::add at /home/vecna/ogre-ws/ogre-1.10.11/OgreMain/src/OgreResourceManager.cpp (line 158)
> terminate called after throwing an instance of 'Ogre::ItemIdentityException'
>   what():  ItemIdentityException: Resource with the name rviz/glsl120/include/pass_depth.vert already exists. in ResourceManager::add at /home/vecna/ogre-ws/ogre-1.10.11/OgreMain/src/OgreResourceManager.cpp (line 158)
> Aborted (core dumped)

I have built my OGRE and rviz with the stereo flags on. Any ideas about how I could resolve this would be much appreciated!

rhaschke, 2019-08-30:
Fixed via #1063.

closed 2019-08-30 by rhaschke