ros-visualization/rviz #1052: can't install rviz

Annaiyan, 2016-10-10:
Hello,

I tried to install rviz but i'm getting the following error:
```
fatal error: urdf_model/types.h: No such file or directory
 #include <urdf_model/types.h>
```
For your information,
I'm using ros-indigo in ubuntu 14.04 trusty.

How to get rid of the above error?

Thanks in advance.

VictorLamoine, 2016-10-12:
The default branch of this repository is for kinetic, try to compile again after switching to indigo-devel branch:
`git checkout indigo-devel`

Annaiyan, 2016-10-12:
I have installed indigo-devel not kinetic .
I have tried couple of times but still the same error exists.

VictorLamoine, 2016-10-12:
I'm saying that you must compile indigo-devel rviz repository if you are using ROS indigo.
By default when you git clone this repository you are on the kinetic-devel branch (of rviz), which is not compatible with ROS indigo (thus the error).

Refer to my first comment to change branch, then compile again.

wjwwood, 2016-10-17:
I agree with @VictorLamoine, I think you need to switch to the `indigo-devel` branch before building. Also, make sure to delete the build folder after switching but before recompiling.

I'm going to close this for now, since I cannot reproduce the issue in any supported configuration. I'd suggest asking on answers.ros.org if you need more help.

closed 2016-10-17 by wjwwood