ros-visualization/rviz #1055: Rviz on jetson tx1 failed!

Abbyls, 2016-10-16:
Hi, 
I want to kown is there any way to use Rviz on jetson TX1(local instead of cross_compile)? As i run the "rviz" on my board, appeared "bus error". And i followed suggestions as "unset GTK_IM_MODULE" in the .bashrc file. Unfortunately, it failed. I really don't kown how to solve it!

wjwwood, 2016-10-17:
Hi @Abbyls, unfortunately this isn't enough information for me to help you, as there are no details like a core file that could tell me why the bus error occurred and I haven't had any experience using rviz on a tx1.

I believe you'll get better help from the answers.ros.org website, as more people watch that site and they have a broader experience with these issues.

I'm going to close this for now. If you find a specific problem or have more details, post them here and I'll reopen the issue.

Thanks.

closed 2016-10-17 by wjwwood

seedjack, 2017-10-09:
Hello!@Abbyls 
I meet the same issue,have you solved it?

Abbyls, 2017-10-14:
No