ros-visualization/rviz #1056: focal point marker can appear blueish

wjwwood, 2016-10-17:
Normally it would be yellow. I can reproduce this in a Xenial VM on my macbook by running `roslaunch navigation_stage move_base_amcl_2.5cm.launch`.

[image]

VictorLamoine, 2016-10-17:
This is expected and not a bug:
https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/default_plugin/view_controllers/third_person_follower_view_controller.cpp#L60

Switch back to `Orbit (rviz)` view controller to get the yellow focal shape.

wjwwood, 2016-10-17:
Ah right, I even remember that now... Thanks.

closed 2016-10-17 by wjwwood