ros-visualization/rviz #1065: OccupancyGrid display ignores timestamps

ayrton04, 2016-11-01:
The OccupancyGrid plugin (map_display) is [transforming](https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/default_plugin/map_display.cpp#L707) the map using a time stamp of `ros::Time()` instead of `current_map_.header.stamp`. If this was intentional, perhaps we can add an option to use the header's time stamp? I'd be happy to submit a PR.

wjwwood, 2016-11-01:
I'm not sure if it was intentional or not. @hersh @DLu can either of you think of a reason the current time was used for the transform in the MapDisplay rather than using the timestamp in the header? Either way, having an option (which defaults to the current behavior) would not do any harm.

NikolausDemmel, 2016-11-02:
When the `OccupancyGrid` represents a grid map, I would assume that `header.stamp` would correspond to the map-creation time, which for the typical localization scenario would lie far in the past and outside the TF buffer, so I think the current behaviour as a default makes sense for that scenario, but I understand that for different uses of OccupancyGrid this might make less sense.

DLu, 2016-11-02:
Nothing comes to mind.

wjwwood, 2017-05-02:
Closed with https://github.com/ros-visualization/rviz/pull/1066

closed 2017-05-02 by wjwwood