ros-visualization/rviz #1067: Bad z-buffer on visualization_msgs::Marker for CUBE_LIST or SPHERE_LIST

m4nh, 2016-11-12:
If you create a CUBE or SPHERE list as Marker type and do a:

```
int N = rand()%10000;
marker.points.resize(N);
marker.colors.resize(N);
for(int i = 0; i < N/2; i++){
 marker.points[i].x = (rand()%1000)/1000.0;
 marker.points[i].y = (rand()%1000)/1000.0;
 marker.points[i].z = (rand()%1000)/1000.0;
 marker.colors[i].r = (rand()%1000)/1000.0;
 marker.colors[i].g = (rand()%1000)/1000.0;
 marker.colors[i].b = (rand()%1000)/1000.0;
 marker.colors[i].a = 1;
}
```

Cubes (or Spheres) now has bad z-buffer. Farthest elements are drawn as first. See attached image, on the left part. (It is also noteworthy that entire voxel map is drawn behind rviz standard grid regardless of its real position that is surely on top of it: **z > 0**)

[image]

Briefly the problem seems to be that when you iterate on a lesser number of points/color lists size something strange happens in z-buffer. The same behavior when you add some overlapping CUBE or SPHERE (two or more elements with same position and same color).
Problem disappears if simple "push_back" is used to fill a void point/color list iteratively.

The bug was tested on several machines with several Video card on: _**RViz version 1.11.15 (indigo).**_

wjwwood, 2016-11-14:
Thanks for the detailed report. I'll have a look as soon as possible. It would be helpful for when someone finds time to look into it to have a simple .bag file and .rviz file (or instructions on how to setup the issue in rviz) so that reproducing the problem is easier.

wjwwood, 2016-11-14:
These issues may be related:

- https://github.com/ros-visualization/rviz/issues/1006
- https://github.com/ros-visualization/rviz/issues/925

ojura, 2017-06-30:
I would like to add that LINE_LIST markers are problematic too:

https://www.youtube.com/watch?v=VVgRpsvlv0Y&feature=youtu.be

This is on Ubuntu 17.04, ROS Lunar.

ojura, 2017-06-30:
@wjwwod here's also a bag and an rviz file. https://www.dropbox.com/sh/ne95xuq11xv1ess/AABIPKOZXnxgKOqeNvlEj4H6a?dl=0

ojura, 2017-07-02:
A small update which could be an useful clue: I changed the following line to be `setDepthWriteEnabled( true )` and this helped prevent the bug: https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/ogre_helpers/billboard_line.cpp#L262

wjwwood, 2017-07-17:
Thanks @ojura, I'll try to have a look at that soon.

ojura, 2017-07-17:
Thanks! By the way, have you reproduced the other issue (>16384 points crash)? Should be fairly simple to reproduce, just fill up a marker with more than 16384 points. But if you need a bag, let me know.