ros-visualization/rviz #1068: Control commands does not work anymore

VictorLamoine, 2016-11-30:
Ubuntu 16.04

```
This is RViz version 1.12.4 (kinetic).

Compiled against Qt version 5.5.1.
Compiled against OGRE version 1.9.0 (Ghadamon)
```

`Ctrl + O` does not open a file dialog.
`Ctrl + S` does not work to save the config file.
`Ctrl + Q` does not quit RViz.

Can anyone reproduce this issue?

davetcoleman, 2017-01-06:
Yes, when I switch back to Indigo I am happy to have Ctrl-S, but in Kinetic a new display/panel/something has stolen that command I think. I would like to restore this behavior.

VictorLamoine, 2017-02-13:
Seems to be solved for me?!

```
[ INFO] [1487009245.278498391]: rviz version 1.12.4
[ INFO] [1487009245.278531194]: compiled against Qt version 5.5.1
[ INFO] [1487009245.278539614]: compiled against OGRE version 1.9.0 (Ghadamon)
```

wjwwood, 2017-05-02:
The hotkeys are working for me on `kinetic-devel`. I'll close this for now. Please comment here if anyone still has the issue and I'll reopen it.

closed 2017-05-02 by wjwwood