ros-visualization/rviz #1069: High DPI issue

boris-il-forte, 2016-12-15:
From ros answers:
http://answers.ros.org/question/249914/rviz-high-dpi-issue/

I'm posting also here because on ros answers I haven't had any informations about this issue.
I don't know if is an issue about my current configuration or an RVIZ issue, unfortunately I don't have another high dpi screen to test the issue on a different configuration.

wjwwood, 2017-04-29:
Can you try `kinetic-devel` now that https://github.com/ros-visualization/rviz/pull/1078 has been merged to see if it fixes your issue?

I'm going to close this and if it is not fixed, just comment here and I'll reopen it.

closed 2017-04-29 by wjwwood

boris-il-forte, 2017-04-29:
This seemes to be already fixed in the version on kinetic repositories. Maybe was an ogre bug?

orzechow, 2018-03-05:
I'm having exactly the same issue with RViz version 1.12.15 (kinetic), compiled against Qt 5.5.1 and OGRE 1.9.0 (Ghadamon)

I'm running ROS on a Linux Mint 18 KDE system (based on Ubuntu 16.04)

One thing I noticed: my system has Qt 5.6.1 installed (python-pyqt5 and python3-pyqt5 5.5.1 though). Maybe this is related?

boris-il-forte, 2018-03-06:
The problem is back again in kubuntu 16.04... Maybe a regression?