ros-visualization/rviz #1070: Tons of Qt warnings when Rviz runs

pbeeson, 2016-12-16:
In the kinetic debian distros, a warning is printed out a lot whenever Rviz starts.  The number of times this warning prints seems to be a function of what is loaded in the .rviz file.  The default (empty) rviz configuration only spews 4 lines, but other spew hundreds. The lines are all the same “Attempt to set a screen on a child window” warning message.
```
[ INFO] [1481909923.838898198]: rviz version 1.12.4
[ INFO] [1481909923.838931162]: compiled against Qt version 5.5.1
[ INFO] [1481909923.838939431]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1481909925.009882314]: Stereo is NOT SUPPORTED
[ INFO] [1481909925.009967339]: OpenGl version: 4.5 (GLSL 4.5).
0x1bd7c10 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x10f7710) ): Attempt to set a screen on a child window.
0x1be10c0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x10f7710) ): Attempt to set a screen on a child window.
0x1bdea90 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x10f7710) ): Attempt to set a screen on a child window.
0x1be59a0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x10f7710) ): Attempt to set a screen on a child window.
```

Apparently, this is due to using a deprecated widget on multi-monitor systems.
http://stackoverflow.com/questions/33545006/qt5-attempt-to-set-a-screen-on-a-child-window-many-runtime-warning-messages

wjwwood, 2017-01-04:
Thanks for the report. I don't have time to look into it at the moment, but I'll get to it eventually. If anyone has time to make pr to fix this it would go faster.

asimay, 2017-10-28:
+1. Run rviz on Kinect version, PX2.
same error occurs. and the rviz will exit and cannot use totally.

asimay, 2017-10-28:
this is urgent request...

wjwwood, 2017-10-31:
I don't have time to look into this right now, but perhaps someone else does (maybe even yourself)?

----

> this is urgent request...

See http://wiki.ros.org/Support#Etiquette:

> - It's considered bad form to list your personal deadlines; Community members answering questions also have them.
> - Do not beg for help. If there is someone willing and able to help with your problem, you usually get a response. Asking for faster answers will mostly have a negative effect.

tlind, 2018-05-30:
+1, seeing the same issue with ROS Kinetic/16.04 on multi-monitor setup

wjwwood, 2018-09-16:
> the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'

Did you try that?

lianghongzhuo, 2020-04-19:
As mentioned [here](https://stackoverflow.com/questions/33545006/qt5-attempt-to-set-a-screen-on-a-child-window-many-runtime-warning-messages):
this warning messages occur whenever `setMinimumWidth()`  ` setMinimumHeight()` are called on a QGLWidget under Linux with a dual screen display.
So I searched in the rviz repo, and find [here](https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/visualization_frame.cpp#L157) called the `setMinimumWidth() ` function.
So a fix for this is as the first link suggested:
> This is probably a bug in Qt. It will probably be not fixed since it is recommended in the documentation to use the new QOpenGLWidget instead, that appeared in Qt 5.4 (note: "OpenGL" instead of "GL"), which I did and the warning messages disappeared.

So "simply" replace `QGLWidget` with `QOpenGLWidget` should work.

rhaschke, 2020-04-20:
I can't reproduce the warning messages in Melodic. In which version / distro do you see them?

lianghongzhuo, 2020-04-20:
Oh, you're right, Melodic version don't have this warning now. Kinectic version, the warning still exists.

rhaschke, 2020-04-20:
We will not fix this issue in Kinetic anymore. Hence, I'm closing this issue.
Thanks for confirming the fix in Melodic.

closed 2020-04-20 by rhaschke