ros-visualization/rviz #1071: rviz crashes on debian testing

nxdefiant, 2017-01-02:
rviz crashes in current debian testing, tested with debian rviz [1.12.4+dfsg-2](https://packages.debian.org/stretch/rviz) and compiled from source git 1cb7c987.

Error is:
```
# rosrun rviz rviz -l                                                                                                                        
[rosrun] You have chosen a non-unique executable, please pick one of the following:
1) /tmp/catkin_ws/devel/lib/rviz/rviz
2) /usr/share/rviz/rviz
#? 1
1483358932.798750578  INFO ${node}: rviz version 1.12.4
1483358932.798832649  INFO ${node}: compiled against Qt version 5.7.1
1483358932.798841999  INFO ${node}: compiled against OGRE version 1.9.0 (Ghadamon)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
1483358933.024578028  INFO /rviz_1483358932798936748: Creating resource group General
1483358933.024645334  INFO /rviz_1483358932798936748: Creating resource group Internal
1483358933.024660708  INFO /rviz_1483358932798936748: Creating resource group Autodetect
1483358933.024706980  INFO /rviz_1483358932798936748: SceneManagerFactory for type 'DefaultSceneManager' registered.
1483358933.024834695  INFO /rviz_1483358932798936748: Registering ResourceManager for type Material
1483358933.024857315  INFO /rviz_1483358932798936748: Registering ResourceManager for type Mesh
1483358933.024873092  INFO /rviz_1483358932798936748: Registering ResourceManager for type Skeleton
1483358933.024892274  INFO /rviz_1483358932798936748: MovableObjectFactory for type 'ParticleSystem' registered.
1483358933.024935970  INFO /rviz_1483358932798936748: ArchiveFactory for archive type FileSystem registered.
1483358933.024950327  INFO /rviz_1483358932798936748: ArchiveFactory for archive type Zip registered.
1483358933.024962032  INFO /rviz_1483358932798936748: ArchiveFactory for archive type EmbeddedZip registered.
1483358933.024973077  INFO /rviz_1483358932798936748: DDS codec registering
1483358933.024990303  INFO /rviz_1483358932798936748: FreeImage version: 3.17.0
1483358933.025004231  INFO /rviz_1483358932798936748: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
zsh: abort      rosrun rviz rviz -l
```

Problem seems to be in the RenderPanel constructor: Ogre and QT are from debian.

```
(gdb) backtrace
#0  0x00007fffed9d3fdf in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007fffed9d540a in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007fffee2eb04d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fffee2e9016 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fffee2e9061 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007fffee2e9279 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fffee31175f in std::__throw_logic_error(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff6d6d6e4 in ?? () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff6d6ee3f in Ogre::FreeImageCodec::startup() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff6c751a5 in Ogre::Root::Root(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff7ad19bf in rviz::RenderSystem::RenderSystem() () from /usr/lib/x86_64-linux-gnu/librviz.so.2d
#11 0x00007ffff7ad1bf5 in rviz::RenderSystem::get() () from /usr/lib/x86_64-linux-gnu/librviz.so.2d
#12 0x00007ffff7acc781 in rviz::QtOgreRenderWindow::QtOgreRenderWindow(QWidget*) () from /usr/lib/x86_64-linux-gnu/librviz.so.2d
#13 0x00007ffff7af3e19 in rviz::RenderPanel::RenderPanel(QWidget*) () from /usr/lib/x86_64-linux-gnu/librviz.so.2d
#14 0x00007ffff7b416c0 in rviz::VisualizationFrame::initialize(QString const&) () from /usr/lib/x86_64-linux-gnu/librviz.so.2d
#15 0x00007ffff7b4d9ef in rviz::VisualizerApp::init(int, char**) () from /usr/lib/x86_64-linux-gnu/librviz.so.2d
#16 0x0000555555555200 in main (argc=<optimized out>, argv=0x7fffffffe528) at /tmp/rviz/ros-rviz-1.12.4.1/src/rviz/main.cpp:40
```

nxdefiant, 2017-01-03:
This seems to be a debian specific bug in [libfreeimage](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=849696), reverting to [libfreeimage 3.17.0+ds1-3](http://snapshot.debian.org/archive/debian/20161105T212628Z/pool/main/f/freeimage/libfreeimage-dev_3.17.0%2Bds1-3_amd64.deb) fixed it.

nxdefiant, 2017-01-03:
Bug [850027](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=850027) filled against debian/libfreeimage. Debian Maintainer wants this to get fixed in rviz.

wjwwood, 2017-01-03:
Yeah, rviz doesn't handle what he's talking about (AFAICS), I think it would have to be fixed in ogre.

wjwwood, 2017-05-02:
I'm going to close this since it needs to be fixed in another project.

closed 2017-05-02 by wjwwood