ros-visualization/rviz #1075: Rviz Segmentation Fault with 3D Acceleration

jmschonfeld, 2017-01-12:
I am currently using VirtualBox with a macOS host and Ubuntu guest OS to run ros/rviz on the Ubuntu VM. When the VirtualBox setting to enable 3D Acceleration within the display settings, rviz encounters a segmentation fault immediately on launching the command. This seg fault does not occur with 3D Acceleration disabled, but disabling this feature causes my VM to run extremely slowly. Any ideas how to run Rviz with 3D acceleration enabled?

wjwwood, 2017-01-28:
I've seen people use VirtualBox with 3D acceleration on use rviz successfully. I've even done it myself on my macbook pro, but not with macOS (not since they changed the name from OS X).

Unfortunately I can't help without more information. If you're looking for others that have managed to make this work recently, then you'd find a large audience on answers.ros.org.

I'm going to close this until there are more details or otherwise new information comes up, at which point I can reopen the issue.

closed 2017-01-28 by wjwwood

DrClick, 2017-09-27:
I have this exact issue with the brand new macbook pro 15 with touchbar. I found the only the worked is using VMWare Fusion 8. Worked the first time, no issues.

dpipemazo, 2017-10-23:
Can confirm same issue. I have some (basic) info that can hopefully get things kicked off on a discussion:

Computer:
Mac OS Sierra 10.12.4

Virtualization:
- VirtualBox 5.2.0 r118431 (Qt5.6.3)
- Ubuntu 16.04
- ROS Kinetic

I can confirm that running the VM without 3D hardware acceleration I'm able to get `rviz` running OK, but with it on ROS prints out the following (when running the tf demo):
```
$ rosrun rviz rviz -d `rospack find turtle_tf`/rviz/turtle_rviz.rviz
[ INFO] [1508775446.492873815]: rviz version 1.12.13
[ INFO] [1508775446.492931257]: compiled against Qt version 5.5.1
[ INFO] [1508775446.492940466]: compiled against OGRE version 1.9.0 (Ghadamon)
Segmentation fault (core dumped)
```

I then ran it through GDB for what it's worth:
```
(gdb) set args /opt/ros/kinetic/bin/rosrun rviz rviz -d `rospack find turtle_tf`/rviz/turtle_rviz.rviz
(gdb) run
Starting program: /bin/bash /opt/ros/kinetic/bin/rosrun rviz rviz -d `rospack find turtle_tf`/rviz/turtle_rviz.rviz
process 2832 is executing new program: /opt/ros/kinetic/lib/rviz/rviz
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffde2aa700 (LWP 2845)]
[New Thread 0x7fffd427d700 (LWP 2846)]
[New Thread 0x7fffd3a7c700 (LWP 2847)]
[New Thread 0x7fffd327b700 (LWP 2848)]
[New Thread 0x7fffd2104700 (LWP 2850)]
[New Thread 0x7fffd1903700 (LWP 2851)]
[ INFO] [1508775728.203901287]: rviz version 1.12.13
[ INFO] [1508775728.203951479]: compiled against Qt version 5.5.1
[ INFO] [1508775728.203960569]: compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffd0ef0700 (LWP 2854)]
[New Thread 0x7fffbbfff700 (LWP 2855)]
[New Thread 0x7fffbb7fe700 (LWP 2856)]
[New Thread 0x7fffbaffd700 (LWP 2861)]
[New Thread 0x7fffb9846700 (LWP 2862)]
[Thread 0x7fffb9846700 (LWP 2862) exited]
[New Thread 0x7fffb9045700 (LWP 2863)]

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) quit
```

Not sure if the above is quite useful, let me know if I can help get more information.