ros-visualization/rviz #1077: Render window 1/4 size with QT5, OSX, Retina display

jonfink, 2017-01-18:
System:

- OSX 10.11
- ROS Kinetic
- QT 5.7.1 (homebrew)
- Ogre 1.9.0 (homebrew osrf/simulation tap)
- RVIZ 1.12.4

After launching Rviz, the render window seems to only draw for the lower-left 1/4 of it's total space:

[image]

I've traced this problem back to be related to [Qt5's "support" for retina displays](https://blog.qt.io/blog/2013/04/25/retina-display-support-for-mac-os-ios-and-x11/) (where the rendering units are scaled).

[Realizing that this problem was recently addressed by the Ogre development team](https://bitbucket.org/osrf/gazebo/issues/2070/rendering-1-4-of-scene-with-qt5-on-macos), I tried the same fix locally (since I am using Ogre 1.9):

In `ogre_helpers/render_system.cpp`
```c++
  params["macAPI"] = "cocoa";
  params["macAPICocoaUseNSView"] = "true";
  // Adding this param for retina display scaling
  params["contentScalingFactor"] = "2.0";
```

Obviously, this fix requires a number of checks (for presence of Qt5 + Ogre 1.9 and check for retina display) before a merge request can be considered but I wanted to report here in case others are hitting the same issue.

folkwang, 2017-03-15:
this happened to me as well on macOS Sierra. My environments are ROS kinetic + Ogre 1.9.0 + rviz version 1.12.4. FYI, Since I am also using an external low DPI display, by dragging rviz window into the external display it would be rendered correctly

wjwwood, 2017-04-29:
Can either of you try `kinetic-devel` now that https://github.com/ros-visualization/rviz/pull/1078 has been merged to see if it fixes your issue?

wjwwood, 2017-04-29:
In fact I'm going to close this optimistically, and if it is not fixed I'll reopen this.

closed 2017-04-29 by wjwwood