ros-visualization/rviz #1081: Wrench visual deadzone

goretkin, 2017-02-15:
I strongly dislike the behavior I'm getting of hiding the arrows if they become "shorter than they are wide". It seems like a good idea, but it should be optional and perhaps a separate parameter should decide the threshold.

It doesn't really seem like it's working as intended:
https://youtu.be/6kV5kS81WsQ
(notice how the deadzone really is too big. the arrows materialize out of nowhere, losing a huge range of sensitivity.)

Though from reading the code, I can't tell why:
https://github.com/ros-visualization/rviz/blob/7970ba08cee3810cfa1609c3b0f5136970eb2f7c/src/rviz/default_plugin/wrench_visual.cpp#L63

v4hn, 2018-02-05:
We just hit this bug too. I would propose to add a boolean flag "hide small forces" for this feature. Would that be ok for you @wjwwood ?

wjwwood, 2018-02-05:
Fine by me.