ros-visualization/rviz #1082: Segfault in Nvidia when using IMs

pbeeson, 2017-02-16:
We've found that when the always_visible() flag is set on markers, and you have attached a mesh to the marker, that Ubuntu nvidia drivers will end up crashing rviz when you are using the controller to change the pose of the marker.. Backtrace of gdb probiced below. This happens on 304--367 Nvidia drivers in Kinetic, but if nouveau is used we don't see it.

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00000000400a1c6e in ?? ()
(gdb) bt
#0  0x00000000400a1c6e in ?? ()
#1  0x00007fffab46d079 in ?? () from /usr/lib/nvidia-367/libnvidia-glcore.so.367.57
#2  0x00007fffab473b44 in ?? () from /usr/lib/nvidia-367/libnvidia-glcore.so.367.57
#3  0x00007fffab05a5a9 in ?? () from /usr/lib/nvidia-367/libnvidia-glcore.so.367.57
#4  0x00007fffb81a89c5 in Ogre::GLRenderSystem::_render(Ogre::RenderOperation const&) ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007ffff3b937f7 in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff3bbdfb9 in Ogre::QueuedRenderableCollection::acceptVisitorGrouped(Ogre::QueuedRenderableVisitor*) const () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff3bbe095 in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff3b7c8df in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff3b7c757 in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff3b92474 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff3bcef51 in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#12 0x00007ffff3b7377a in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff3b735cb in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff3b7311e in Ogre::RenderTarget::updateImpl() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff3b73ac4 in Ogre::RenderTarget::update(bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff3c23e5f in Ogre::RenderSystem::_updateAllRenderTargets(bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff3d3a4ba in Ogre::Root::_updateAllRenderTargets() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff3d3a5a0 in Ogre::Root::renderOneFrame() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#19 0x00007ffff7b4d3ce in rviz::VisualizationManager::onUpdate() ()
   from /opt/ros/kinetic/lib/librviz.so
#20 0x00007ffff2193d2a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff21a05c8 in QTimer::timerEvent(QTimerEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff2194bb3 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff746a05c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff746f516 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff216538b in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff21ba5ed in QTimerInfoList::activateTimers() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff21baaf1 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff16bc197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff16bc3f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff16bc49c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff21bb7cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
---Type <return> to continue, or q <return> to quit---q
 (Quit
(gdb) q

romainreignier, 2017-03-09:
Same problem here with Nvidia 378 driver.
Crash when on a click on `Publish Point` in the config of [this tutorial](http://wiki.ros.org/teb_local_planner/Tutorials/Following%20the%20Global%20Plan%20%28Via-Points%29).

```
Thread 8 "rviz" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc99e5700 (LWP 22638)]
0x00000000400a1c8a in ?? ()
(gdb) bt
#0  0x00000000400a1c8a in ?? ()
#1  0x00007fffbf404ce4 in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#2  0x00007fffbefde378 in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#3  0x00007fffbefce6ff in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#4  0x00007fffbefd509d in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#5  0x00007fffc9c9d12c in ?? () from /usr/lib/nvidia-378/libGLX_nvidia.so.0
#6  0x00007ffff5ed86ba in start_thread (arg=0x7fffc99e5700) at pthread_create.c:333
#7  0x00007ffff6ab482d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb)
```

pbeeson, 2017-03-09:
@romainreignier Please try adding the following to your .bashrc and resourcing it.  The solution below seems to work for pthread type issues, but didn't solve my problem which always originates from an Ogre render_() call.

```
##############
export LD_PRELOAD=/lib/x86_64-linux-gnu/libpthread.so.0
### From HERE: http://git.net/ml/ubuntu-bugs/2016-06/msg09689.html
## Works for Nvidia segfault on glmark2
```

romainreignier, 2017-03-09:
@pbeeson Thanks for your quick answer. I will try that solution as soon as I reboot. But for now, I have switched to my Intel card.

romainreignier, 2017-03-13:
@pbeeson I have tried your solution but it seems it does not change anything:

```
Starting program: /opt/ros/kinetic/bin/rviz -d cfg/rviz_test_optim.rviz
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdec29700 (LWP 7063)]
[ INFO] [1489393298.031555975]: rviz version 1.12.4
[ INFO] [1489393298.031616014]: compiled against Qt version 5.5.1
[ INFO] [1489393298.031630185]: compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffd2507700 (LWP 7080)]
[New Thread 0x7fffd1d06700 (LWP 7081)]
[New Thread 0x7fffd1505700 (LWP 7082)]
[New Thread 0x7fffd0d04700 (LWP 7087)]
[New Thread 0x7fffc99e5700 (LWP 7104)]
[Thread 0x7fffc99e5700 (LWP 7104) exited]
[New Thread 0x7fffc99e5700 (LWP 7105)]
[New Thread 0x7fffbd413700 (LWP 7158)]
[New Thread 0x7fffbcc12700 (LWP 7159)]
[New Thread 0x7fffb7fff700 (LWP 7160)]
[New Thread 0x7fffb77fe700 (LWP 7161)]
[New Thread 0x7fffb6ffd700 (LWP 7162)]
[New Thread 0x7fffb67fc700 (LWP 7163)]
[New Thread 0x7fffb5ffb700 (LWP 7164)]
[New Thread 0x7fffb57fa700 (LWP 7165)]
[ INFO] [1489393298.894351059]: Stereo is NOT SUPPORTED
[ INFO] [1489393298.894464800]: OpenGl version: 4.5 (GLSL 4.5).
[New Thread 0x7fffb4ff9700 (LWP 7180)]
[New Thread 0x7fff8bfff700 (LWP 7181)]
[New Thread 0x7fff8ad60700 (LWP 7223)]

Thread 8 "rviz" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc99e5700 (LWP 7105)]
0x00000000400a1c8a in ?? ()
(gdb) bt
#0  0x00000000400a1c8a in ?? ()
#1  0x00007fffbf404ce4 in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#2  0x00007fffbefde378 in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#3  0x00007fffbefce6ff in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#4  0x00007fffbefd509d in ?? () from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#5  0x00007fffc9c9d12c in ?? () from /usr/lib/nvidia-378/libGLX_nvidia.so.0
#6  0x00007ffff7bc16ba in start_thread (arg=0x7fffc99e5700) at pthread_create.c:333
#7  0x00007ffff689782d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

wjwwood, 2017-05-01:
I can produce this with the teb local planner tutorial. I'm using 375.39.

wjwwood, 2017-05-01:
I don't get the traceback @romainreignier is talking about, but I do get the original one @pbeeson and @ompugao are talking about.

wjwwood, 2017-05-02:
Ok, I looked into this for a while and I wasn't able to find the problem easily. Here are some notes though:

- You do not need to click with the publish point tool to trigger the segfault
  - You can just select the tool and move your mouse around in the viewport
- You can remove everything except the `get3DPoint()` and it will still segfault, but if you remove that call in the `PointTool::processMouseEvent()` method it will not crash
  - https://github.com/ros-visualization/rviz/blob/7970ba08cee3810cfa1609c3b0f5136970eb2f7c/src/rviz/default_plugin/tools/point_tool.cpp#L98
- The segfault occurs in thread 1, which is the Qt main thread and from the `onUpdate()` method, which ends up calling the Ogre render function
  - However, the code that causes the segfault is also executed in this thread, therefore I do not think it is a race condition, but I could be wrong

I wasn't able to find any Ogre or Qt or nvidia threads about this, or none that I think are related.

pbeeson, 2017-05-02:
Thanks for spending some time on this. Those of us working on object based
manipulation use IMs a lot and have been dealing with this for some time.

One non-proven bit of info I can share is that by removing alpha-ed meshed
surrounding the current IM that is being examined, the probability of
crashes goes way down. So I see it much more in complex, crowded scenes
than in simple ones.

Willyzw, 2017-05-11:
I got exactly same backtrace as @pbeeson. Each time I click on an interactive marker and want to render it again, the segfault error occurred. Here I use ubuntu 16.04 with Nvidia driver 375.26 and ROS kinetic.

Interestingly if there are something behind the marker, then it is fine,  I can click it and render it again as many time as I want..

Jmeyer1292, 2017-10-01:
I too get a similar backtrace as @pbeeson. Ubuntu 16.04.3 (default desktop), Nvidia 384.81, ROS Kinetic.

I planned on using interactive markers and the publish points tool with a CUDA program so this is problematic.

pbeeson, 2017-12-28:
It is possible this was fixed by [PR1167](https://github.com/ros-visualization/rviz/pull/1167).  We had been using a pretty old library for setting up IM controls that had quaternions of magnitude 2.  I never realized this until the latest Rviz stopped displaying our IM controls.

pbeeson, 2018-09-27:
Update: This was not fixed by PR1167.  I'm still getting this using the latest 16.04.05

```(gdb) bt
#0  0x00000000400a2c6e in ?? ()
#1  0x00007fffed2e5e59 in ?? () from /usr/lib/nvidia-396/libnvidia-glcore.so.396.44
#2  0x00007fffed2ecae4 in ?? () from /usr/lib/nvidia-396/libnvidia-glcore.so.396.44
#3  0x00007fffeceaf0a5 in ?? () from /usr/lib/nvidia-396/libnvidia-glcore.so.396.44
#4  0x00007fffb27099c5 in Ogre::GLRenderSystem::_render(Ogre::RenderOperation const&) () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007ffff39597f7 in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff3983fb9 in Ogre::QueuedRenderableCollection::acceptVisitorGrouped(Ogre::QueuedRenderableVisitor*) const () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff3984095 in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff39428df in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff3942757 in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff3958474 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff3994f51 in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#12 0x00007ffff393977a in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff39395cb in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff393911e in Ogre::RenderTarget::updateImpl() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff3939ac4 in Ogre::RenderTarget::update(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff39e9e5f in Ogre::RenderSystem::_updateAllRenderTargets(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff3b004ba in Ogre::Root::_updateAllRenderTargets() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff3b005a0 in Ogre::Root::renderOneFrame() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#19 0x00007ffff7b4c18e in rviz::VisualizationManager::onUpdate() () from /opt/ros/kinetic/lib/librviz.so
#20 0x00007ffff1eaed2a in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff1ebb5c8 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff1eafbb3 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff746405c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff7469516 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff1e8038b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff1ed55ed in QTimerInfoList::activateTimers() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff1ed5af1 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff13d7197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff13d73f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff13d749c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff1ed67cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff1e7db4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007ffff1e85bec in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x0000000000400c99 in main ()```

pbeeson, 2018-09-27:
I'm wondering if rviz::VisualizationManager::onUpdate() could have a simple catch on Ogre::Root::renderOneFrame(), if maybe bringing down all of rviz might be possible.

pbeeson, 2018-10-06:
I’ve noticed that I can determinisitcally make this fail by having a marker array drawn and then touching an Interactive Marker right click pulldown menu, so I’m still not convinced this isn’t some memory (indexing) issue originating in Rviz.

awatson3, 2018-11-06:
I can confirm that we're also seeing RViz segfault like pbeeson while using interactive markers in RViz.  We are running kinetic, xenial, and nvidia drivers.

pbeeson, 2018-11-06:
I have discovered that this is easily reproducible when sending a marker
array of size 1000 then right clicking on any IM. I will try to post a
simple deterministic failure soon.

d-walsh, 2019-01-09:
I had this problem but I believe I was able to overcome it by installing [OGRE v1.9.1](https://github.com/OGRECave/ogre/releases/tag/v1.9.1) from source. Followed by recompiling Rviz from source.

GMahmoud, 2019-01-29:
Hello guys !!
I am having the same issue when a line strip intersects an interactive marker.
Does anybody have a fix ?

GMahmoud, 2019-01-29:
@d-walsh Do you have an idea why installing OGRE and RviZ from source fixes this issue ?

jacobhuesman, 2019-04-24:
Had the same issue trying to use the Publish Point tool. Installing OGRE v1.9.1 and recompiling RViz from source fixed the issue for me. Thanks @d-walsh

coreZair, 2019-04-30:
If this is working for real, could someone in a mighty position update the official ros-kinetic-rviz package in the ROS' ubuntu package sources for xenial (kinetic), bionic (melodic) and the other versions?
Would be helpful for all of us!

wjwwood, 2019-05-06:
We use the ogre from Ubuntu in ROS Kinetic/Melodic, so we cannot change the Ogre version, sorry. It sounds like building from source is the best bet.

martinmeinke, 2019-05-13:
> I had this problem but I believe I was able to overcome it by installing [OGRE v1.9.1](https://github.com/OGRECave/ogre/releases/tag/v1.9.1) from source. Followed by recompiling Rviz from source.
> 
> This might also help for #1192 and #1185.
When I am installing ogre 1.9.1 release source, the libs still carry the 1.9.0 version number, is that expected?

jacobhuesman, 2019-05-13:
Are you sure you're running the recompiled version and not the one that came with ROS?

martinmeinke, 2019-05-14:
@jacobhuesman Yes, I am certain about that! (when I build rviz against any other version of ogre, cmake shows me the version number of the located package)

rhaschke, 2019-09-23:
I can confirm that this bug disappears with Ogre 1.9.1. Hence, closing this issue here.
I filed a [bug report on launchpad](https://bugs.launchpad.net/ubuntu/+source/ogre-1.9/+bug/1845027), asking for an upgrade of Ogre in Ubuntu Bionic and Xenial. There is no other option to fix this.
People affected by this bug should build Ogre 1.9.1 and rviz from source.

closed 2019-09-23 by rhaschke

dave992, 2019-11-29:
I am currently experiencing a similar issue where an segmentation fault occurs in Rviz on ROS-Melodic, however my computer is not equipped with a Nvidia card. The application also uses Interactive Markers. I've included the backtrace:
```
#0  0x00007fffdfb2546c in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#1  0x00007fffdfb25507 in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#2  0x00007fffdfb2730e in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#3  0x00007ffff187a576 in QRasterPaintEngine::drawCachedGlyphs(int, unsigned int const*, QFixedPoint const*, QFontEngine*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#4  0x00007ffff187e1e8 in QRasterPaintEngine::drawTextItem(QPointF const&, QTextItem const&) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#5  0x00007ffff1899b64 in  () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#6  0x00007ffff170af43 in QTextLine::draw(QPainter*, QPointF const&, QTextLayout::FormatRange const*) const () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#7  0x00007ffff170ba25 in QTextLayout::draw(QPainter*, QPointF const&, QVector<QTextLayout::FormatRange> const&, QRectF const&) const () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#8  0x00007ffff73efe9e in QWidgetLineControl::draw(QPainter*, QPoint const&, QRect const&, int) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff73e6055 in QLineEdit::paintEvent(QPaintEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff72e2048 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff73ea9b2 in QLineEdit::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff72a383c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff72ab104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff6c8e9c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff72db19a in QWidgetPrivate::sendPaintEvent(QRegion const&) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff72db75a in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff72dc3c0 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff72db366 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff72dc3c0 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#30 0x00007ffff72dc549 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets [... cut ...]
#68 0x00007ffff00236dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#69 0x00007ffff6ce797f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#70 0x00007ffff6c8c9fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#71 0x00007ffff6c95aa4 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#72 0x0000555555554dc9 in main ()
```
Is this also related to my Ogre version?

rhaschke, 2019-11-29:
At least the segfault occured not in the ogre library, but in plain qt.
If you can reliably reproduce this error, a more detailed description of your environment (or even better a docker) is needed to reproduce and tackle the issue.

dave992, 2019-12-02:
Thanks for the response, the used environment:
ROS-Melodic + ROS-I ScanNPlan (godel) [fork](https://github.com/sam-xl/godel)

To reproduce the segfault I just have to stay in the affected GUI window for 10+ seconds, when I then try to get focus or click something in Rviz it crashes. How long it takes differs but is generally speaking not longer than 1 minute. I will try to make a docker of the environment today/tomorrow in which I can reproduce the issue.