ros-visualization/rviz #1085: Rviz and use_sim_time

ksatyaki, 2017-03-08:
I am running a simulation on gazebo and using rviz for visualization.

I have transforms between different frames available on the /tf topic. The time stamps are in simulated time. use_sim_time is on. Clock is published by gazebo.

Then, I publish certain visualization markers.

The problem is that if the visualization markers are in frame that is not the fixed frame in RViz, RViz tries to find the transform at the current Wall time despite the fact that 'use_sim_time' is true. This fails horribly, of course, and results in a tf extrapolation-into-the-future error. The RViz clock panel recognizes this fact and displays the time that is published on the /clock topic. However, if the marker is in the the world frame (or the fixed frame in RViz), RViz has no problems displaying them. This is expected since the TransformListener is not used.

Thanks.

Note: copy pasted from [here](http://answers.ros.org/question/255910/rviz-and-use_sim_time/)

wjwwood, 2017-05-01:
It might be a bug in the visualization marker display. I don't have time to look into this atm, but a pr would be appreciated!

v01d, 2018-09-12:
hi, i'm wondering if this may be the reason for a similar problem I have: if I publish a cloud w.r.t. frame B, set A as fixed_frame and continuously publish a transform from B to A, the point cloud does not move. I have to republish to be in sync with the transform.
Where in the code did you found this problem?

rhaschke, 2019-09-24:
@v01d, your issue is unrelated to this one. You are essentially expecting a visualization to be updated if a new TF transform is received. However, this is not intended to happen: The TF frame is resolved only once when a message is received. Note that TF frames have a timestamp as well as visualization items usually have: rviz displays the item in the pose at that given timestamp. Thus there is no need to update the pose.
The only exception I know of are _frame_locked_ markers: They get updated if the TF frame changes over time.