ros-visualization/rviz #1086: Slow Rendering of High-Poly Models

Jmeyer1292, 2017-03-13:
I'm stuck on a performance issue related to RVIZ and I'd like to at least ask if anyone has experience with something similar.

I've put together a desktop machine and am facing a "strange" performance bug.

I run the following command to visualize a set of models from our open source robot blending project:
https://github.com/ros-industrial-consortium/godel/blob/kinetic-devel/godel_robots/abb/godel_irb2400/godel_irb2400_support/urdf/irb2400_workspace.xacro

closed 2017-03-13 by Jmeyer1292

wjwwood, 2017-03-14:
@Jmeyer1292 so, did you fix it or why did you close it?

Jmeyer1292, 2017-03-14:
I did the fix the issue, but I'm not entirely sure how. Some combination of blacklisting `nouveau` (which wasn't done automatically?) and changing driver versions made it work. I'm not sure exactly what I did.

I apologize for bothering you.

wjwwood, 2017-03-14:
No worries, just didn't know what the final state was 😄.