ros-visualization/rviz #1087: Path color not saved in config file

theonteg, 2017-03-19:
As described in the topic of the issue, rviz does not save Path lines color to the config file.

As a result every time i load a config, path lines are the same pink default color.

Is this a bug?

wjwwood, 2017-03-21:
It sounds like a bug. Unfortunately I don't have time to verify it and/or look into a fix for it.

You might start by looking at the config's section for the path display and see if there is a field like that in there. If not, then you might need to modify the display plugin for path's to store and use it. Otherwise it might just be a bug in the path display plugin. Either way, you can see the code for it here:

- https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/default_plugin/path_display.h
- https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/default_plugin/path_display.cpp

hiddewie, 2017-03-23:
I've just today also run into this bug.

I looked into the code, and it seems related to the fact that in `PathDisplay` *two* properties are defined with the name `Color`. See https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/default_plugin/path_display.cpp#L99 and https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/default_plugin/path_display.cpp#L71. 

A solution would probably be renaming the pose color as this property should have a different name.

It would be nice to test this when adding properties to a certain display type to check if a property already exists with that name.

romainreignier, 2017-03-23:
@hiddewie Thanks for taking the time to fix this issue.
I have faced that issue for several weeks but did not take the time to track the issue and find a fix.
:+1:

closed 2017-05-01 by wjwwood