ros-visualization/rviz #1091: Wrong work with large coordinates

CodeGenerator, 2017-04-26:
Hello!
I changed in rviz example http://wiki.ros.org/rviz/Tutorials/Markers%3A%20Basic%20Shapes position of the marker object to
```
marker.pose.position.x = 4E6;
marker.pose.position.y = 6E6;
marker.pose.position.z = 0;
```
After that I opened rviz and changed origin to (4E6, 6E6, 0). If I move camera around marker (especially when object type is `visualization_msgs::Marker::ARROW`) I see that arrow breaks up on several parts. Moreover for some angles the marker blinks and disappears.

Do we have some way to allow rviz work with large coordinates without such glitches?

wjwwood, 2017-04-26:
I think this is caused by issues with the underlying Ogre3D implementation of some of the marker types, especially line lists and arrow. There are some issues related to this already:

- https://github.com/ros-visualization/rviz/issues/987
- https://github.com/ros-visualization/rviz/issues/502
- http://www.ogre3d.org/forums/viewtopic.php?f=3&t=68535

So the short answer is no. Typically what people do is pick an arbitrary origin nearby and use that as the fixed frame. It's uncommon that you want to look at two things simultaneously which are separated by such great distances.

If you figure out a more general solution, we'd be happy to try and integrate it into rviz via a pull request.

CodeGenerator, 2017-05-02:
Hello again!
Accordingly to my investigation this problem occurs because rviz uses `float32` for object position manipulations. I recompiled ogre engine with double precision (`OGRE_CONFIG_DOUBLE`) and compiled rviz with that Ogre instead of system one, made few corrections where `float` was used instead of `Ogre::Real`. After that aforementioned example started to work well without degradations!
I tried to go a little further and test other markers but found that complex markers like `SPHERE_LIST` use custom shaders and stuck there. There is constructor `PointCloudRenderable::PointCloudRenderable` where a shader is built by definition. If I change `Ogre::VET_FLOAT3` to `Ogre::VET_DOUBLE3` for position slot in shader and make correspondent changes in `PointCloud::addPoints` I get Ogre lock vertex buffer errors and suggestions to recreate buffer.
Could you assist me how I should change `PointCloud` to make it work with `double` type for position?

wjwwood, 2017-05-02:
@CodeGenerator that's great news that you've made progress. I'd like to help you out, but could you post links to the code you're talking about? I might be able to make recommendations if I'm certain which parts you're having trouble changing to doubles.

CodeGenerator, 2017-05-03:
I pushed my changes here https://github.com/CodeGenerator/rviz/tree/feature/float64

CodeGenerator, 2017-05-05:
What can you suggest to try? Do you know why type substitution didn't work for shader?

wjwwood, 2017-05-08:
@CodeGenerator I do not. Unfortunately I'd have to spend more time than I have digging into the issue to be really helpful. But since you've made the issue more visible, perhaps someone else in the community could help you with it too.