ros-visualization/rviz #1103: rviz can not work and catkin_make error

zhehangT, 2017-05-13:
Ubuntu: 16.04
ROS:kinetic

Today when I startup rviz, it can not work as #1047. 
[image]

It seems to be the qt version incompatible as rviz compiled against **Qt version 5.5.1** while the Qt version in my ubuntu is Qt version 5.6.1. However I not sure.
[image]
[image]

I try to fix the rviz by compiling it in my catkin workspace. But it fails again because of the catkin_make error. 
yaml-cpp I am using is 0.5.3.

[image]

I don't know how to continue fix the rviz.

wjwwood, 2017-05-13:
Please do not post screenshots of text. I cannot be searched or copy-pasted.

Please make sure you have done `sudo apt update` and then `sudo apt install ros-kinetic-rviz`. My machine is using the right versions of qt. Also I can build rviz (branch `kinetic-devel`) from source too. It seems like the only possibility is that your machine has the wrong versions of dependencies. You might also check to make sure you do not have extra PPA's that are pulling in newer versions of libraries like yaml-cpp and qt.

zhehangT, 2017-05-13:
@wjwwood Sorry for posting screenshots of text.

I am wondering what version of yaml-cpp and qt  in your machine.
Thank you.

zhehangT, 2017-05-14:
I compile rviz in my catkin workspace but rviz can not work as well. Just like #1047.

Don not know how to fix it. Sad.

closed 2017-05-14 by zhehangT

zhehangT, 2017-05-15:
I remove the yaml-cpp I installed manual from the github, rviz can wok well.
Maybe the conflict of yaml-cpp version cause the crash

wjwwood, 2017-05-15:
> I remove the yaml-cpp I installed manual from the github

Yup.

> I am wondering what version of yaml-cpp and qt in your machine.

I have Qt `5.5.1` and yaml-cpp `0.5.2-3`.

bigbellmercy, 2021-02-22:
In my case, the errors like the title of this issue disappeared after `rviz` source code reinstalled from its GitHub site.
My environment: Ubuntu 18.04, caktin build (not cakin_make), ROS Melodic