ros-visualization/rviz #1107: Floating point exception in billboard path display when path size greater than 16384

jacobperron, 2017-05-24:
To reproduce:
    - Have a node publish a path message greater of length 16385 or greater.
    - Attempt to view the path in RViz with Line Style set to "Billboards".
    - RViz crashes with floating point exception.

Appears to display fine with path sizes <= 16384.

Possibly related: #957

ojura, 2017-07-02:
Also happens with LINE_STRIP markers.

wjwwood, 2017-07-12:
@jacobperron, or @ojura, if you have a bag file or Python script that can produce a sample path which causes the problem, that would save me some time in debugging it.

wjwwood, 2017-07-17:
In reference to https://github.com/ros-visualization/rviz/issues/1067#issuecomment-315837689:

I have not, I'm trying to work through my backlog as fast as I can. If you don't provide a Python script or bag file then I'll still get to asap, but it always helps to have something from the reporting party to make sure I'm not making some assumptions you're not, which might end up being important.

jacobperron, 2017-07-26:
@wjwwood Here is a bag containing a long path: https://drive.google.com/open?id=0B9O4b91VYXMdNWpCcTdzejUxMk0

Viewing as *Line* is good, but viewing as *Billboards* results in crash.

wjwwood, 2017-07-26:
Thanks @jacobperron I'll have a look as soon as possible.

ojura, 2017-07-27:
Here's a bag and an rviz config for the LINE_STRIP marker.
https://www.dropbox.com/s/oa3jlijzrvygj9p/rviz_linestrip_crash.zip?dl=0

wjwwood, 2017-07-27:
So, I figured out how to make it not crash, see: https://github.com/ros-visualization/rviz/tree/fix_large_billboard_line_chain

But I'm not sure that's the correct solution, or put another way, I'm not sure why there is a `MAX_ELEMENTS` and I'm not sure defaulting `lines_per_chain_` to `1` when drops below that is the right thing to do. Since I don't know what it should look like rendering, I don't know if this change makes it not crash but then doesn't render it correctly.

I don't know what I'm going to do to try and test it out. I could find a regular shape, which I can predict what it should look like and then see if it renders correctly with the change. But my next step is to just try and understand why the limit is hardcoded in there and try to understand what exceeding it (resulting in a value between 1 and 0) actually means.

jacobperron, 2017-07-27:
To test rendering correctness, could you not just duplicate the path topic where one is visualized as lines (presumably correct) and the other as billboards? They should overlay each other perfectly.

wjwwood, 2017-07-27:
@jacobperron that's a good idea, though it might be hard to see when the set is so big. I think I'd basically have to zoom in and pan over the entire thing. When you zoom way out you get some strange rendering artifacts with the billboards, but I don't there's anything to be done about that.

ojura, 2017-08-25:
> when drops below that

How can `lines_per_chain_` be less than zero? Sounds like there's an integer overflow somewhere in the billboard pipeline, and having more than 16384 points results in some calculation turning out a negative result.

wjwwood, 2017-08-25:
@ojura I don't think `lines_per_chain_` is becoming less than zero, but I think it was becoming zero and I think that is because `MAX_ELEMENTS / max_points_per_line_` is integer division and if `max_points_per_line_` is ever greater than `MAX_ELEMENTS` the result will be zero, e.g. `1 / 2 = 0` in integer division. Since this is a division operation, I think it is unlikely that there is an integer overflow or underflow. Also the set we're talking about is nowhere near the signed or unsigned capacity of a 32-bit integer.

ojura, 2017-08-26:
Yes, I was reading your words too literally. I have tried running git blame to see who introduced the `MAX_ELEMENTS`, but the line originates from an earlier SVN repo (https://code.ros.org/svn/ros-pkg/stacks/visualization_common), which is offline.

MatthiasNieuwenhuisen, 2018-06-21:
The proposed workaround results in a crash of RViz caused by too many elements.
```rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreBillboardChain.cpp:673: virtual void Ogre::BillboardChain::updateIndexBuffer(): Assertion `((e + seg.start) * 2) < 65536 && "Too many elements!"' failed.```

The bag file used for testing containing a path with 148,520 points is attached.

[2018-06-21-12-36-27.bag.zip](https://github.com/ros-visualization/rviz/files/2123255/2018-06-21-12-36-27.bag.zip)

wjwwood, 2018-07-11:
@MatthiasNieuwenhuisen thanks for the sample bag file. I won't have time to look into this, though others might pick it up out of the backlog.

goldbattle, 2020-05-31:
This is still an issue in melodic for me, I get the following:
`rviz crashed with SIGFPE in rviz::BillboardLine::setupChains()`

I fixed it on my application side by downsampling the path once it got long:
```
std::vector<geometry_msgs::PoseStamped> poses_imu;
nav_msgs::Path path_arr;
for(size_t i=0; i<poses_imu.size(); i+=std::floor(poses_imu.size()/16384.0)+1) {
    path_arr.poses.push_back(poses_imu.at(i));
}
```

msakuta, 2020-08-25:
This happens here http://docs.ros.org/jade/api/rviz/html/c++/billboard__line_8cpp_source.html#l00151
```
uint32_t lines_left = num_lines_ % lines_per_chain_;
```
If `lines_per_chain_` is zero the modulo will become division by zero.

rhaschke, 2021-11-06:
Fixed via #1662.

PR #1662 (2021-09-24): BillboardLine: Fix handling of many points (> 16384)
For a small number of total points, we place several lines into a single `Ogre::BillboardChain`.
However, if we have more than `MAX_ELEMENTS` points in a line, we need to distribute a line across several BillBoardChains.
Fixes #1107
closed 2021-11-06 by rhaschke