ros-visualization/rviz #1116: RVIZ open config and save config shortcuts not working

shahvi, 2017-06-22:
Hi folks,
I observed this on my installation and was wondering if anyone else has seen it too.
In RVIZ, under the file menu, keyboard shortcuts 'Ctrl+O' and 'Ctrl+S' are offered for 'Open Config' and 'Save Config' respectively.  However, I am unable to use either of these shortcuts.  Has anyone else experienced this?
ROS version: Kinetic
OS: Ubuntu 16.04

Cheers,
Vik

Kukanani, 2017-06-28:
I can confirm that this is an issue on my machine. For example, when pressing Ctrl+S, the Save action does not occur, but instead, the Select tool is activated. Ctrl+O does nothing instead of opening Open.

I have a fix in the works: #95

dhood, 2017-08-22:
@shahvi this should have been addressed by #1117 (thanks @Kukanani), which has been released now. If you don't see the issue's fixed after updating rviz to >=1.12.11 please re-open.

dhood, 2017-08-22:
correction: if it's not fixed please comment here and we'll re-open on your behalf

closed 2017-08-22 by dhood