ros-visualization/rviz #1120: Visualization markers ignore color and show everything as white

gaya-, 2017-07-06:
After the last update on Indigo visualization markers started to appear white.
The message is the same as before and has correct color data.
I suspect it was one of the latest commits about collada file visualization.
Do I understand it correctly that the color information from the Collada file has now higher precedence than the visualization marker message color? If so, what is the reasoning behind that?

wjwwood, 2017-07-11:
@gaya- sorry for the regression. That code has been tricky, I'll have a look and see if I can just revert those changes for now and do another release.

wjwwood, 2017-07-12:
@gaya- I'm guessing it is this pr: https://github.com/ros-visualization/rviz/pull/1084

Can you provide a script or bag file (with mesh file) which causes this problem for you?

wjwwood, 2017-07-12:
Cross-reference to the answers.ros.org: http://answers.ros.org/question/265425/ros-indigo-rviz-maker-turned-white-after-update/

gaya-, 2017-07-14:
Unfortunately the cad models that created this problem are proprietary so I cannot share them :/
And we have a big deadline in 2 weeks, so no time to look deeper into this.
If no one else comes up with a test case I can look into this after the deadline and see what makes our meshes so special and maybe create a similar mesh to test...

wjwwood, 2017-07-25:
I just spent some time trying to reproduce this with Indigo, but I cannot. The behavior in https://github.com/ros-visualization/rviz/pull/1084 seems correct to me and seems to still be the case with the latest version of rviz for Indigo. (btw @gaya- exactly which version of rviz are you using?)

I'm going to wait until either @gaya- or someone else can provide me with a bag file or simple Python script to reproduce it.

HenrySleek, 2017-07-31:
Hey! I have the very same problem.
I am simply using this code:
```
marker.header.frame_id = "base_footprint";
    marker.header.stamp = ros::Time::now();
    marker.ns = "basic_shapes";
    marker.id = 0;
    marker.type = visualization_msgs::Marker::MESH_RESOURCE;
    marker.action = visualization_msgs::Marker::ADD;
    marker.mesh_resource = "package://autonomous_grasping/mesh/Gripper_Assy.stl";
    marker.mesh_use_embedded_materials = false;
    marker.color.r = 0.4;
    marker.color.g = 0.4;
    marker.color.b = 0.4;
    marker.color.a = 1.0;
    marker.scale.x = 0.0012;
    marker.scale.y = 0.0012;
    marker.scale.z = 0.0012;
```
Which I believe is the normal code for spawning a mesh as Marker (the position is separately set). Still, it appears white, and it didn't until a month ago. It is white with any .stl model I tried. It is gray (as I want it) sometimes, like when I run it after restarting the pc, but then it goes back to white. This thing is driving me crazy.

wjwwood, 2017-07-31:
@HenrySleek what version of rviz are you using?

Unfortunately your snippet isn't very helpful without the mesh and the rest of the surrounding code. I'll try to use it as a template to test this out. I was testing with dae before.

wjwwood, 2017-07-31:
Ok, I can reproduce this, but only on Kinetic (Xenial). I can also fix by (re?)applying #1084 to kinetic-devel. I was trying Indigo before, but I think the patch that fixes the issue is already present on the latest version of Indigo:

https://github.com/ros-visualization/rviz/blob/d9b806daceee0596c008289df6252bfdd494c5d3/src/rviz/default_plugin/markers/mesh_resource_marker.cpp#L194-L215

@gaya-, I'll ask again which version of rviz exactly are you using on Indigo?

wjwwood, 2017-07-31:
I merged #1129, so if anyone with this problem on Kinetic could try it out that would be great.

I'm still looking into Indigo's issues.

gaya-, 2017-08-01:
I'm using version 1.11.16.

Here is a sample python code with a sample collada mesh:
  https://github.com/gaya-/marker_test
The mesh was exported from blender.

And here's a screenshot of rviz:
  https://github.com/gaya-/marker_test/blob/master/marker.png

You run the script like this:
`rosrun marker_test marker_test.py`

Running Ubuntu 14.04, ROS Indigo, GeForce GTX 980 graphics card, NVidia driver, nothing all that special.

wjwwood, 2017-08-03:
Ok, I've narrowed down the issue. There must be a problem with the code that sets the color initially. @gaya- if you start rviz, start your example, visualize it in rviz (it will be white), then edit your example to change the color or alpha, then run it again (without closing rviz) then it will render correctly.

I'm still tracking down the source of logic error, but at least I've narrowed down/reproduced the issue.

wjwwood, 2017-08-04:
@gaya- and @HenrySleek can any of you try the patch in https://github.com/ros-visualization/rviz/pull/1132?

gaya-, 2017-08-07:
The bug is gone in version 1.11.17 of indigo-devel.
Thanks a lot for looking into this.

ejalaa12, 2018-01-23:
Hello, I have the same issue. When I try to use a collada file it appears as white. I saw various Issues and PRs #751 #1129 #1084 . And I still don't get it.
I am using ROS Kinetic 1.12.12, and Rviz 1.12.15.
Here is the code snippet that I tried:
```python
#!/usr/bin/env python
import rospy
from std_msgs.msg import ColorRGBA
from visualization_msgs.msg import Marker, MarkerArray
from geometry_msgs.msg import Point, Vector3, Quaternion

def createMarker(frame, mtype, m_id, position=Point(0, 0, 0), orientation=Quaternion(0, 0, 0, 1),
                 scale=Vector3(1, 1, 1), rgba=None):
    """
    :rtype: Marker
    :return:
    """
    marker = Marker()
    marker.header.frame_id = frame
    marker.type = mtype
    marker.id = m_id
    marker.action = Marker.MODIFY

    marker.pose.position = position
    marker.pose.orientation = orientation
    marker.scale = scale
    if rgba:
        r, g, b, a = rgba
        r, g, b = r / 256., g / 256., b / 256.
    else:
        r, g, b, a = 0, 0, 0, 0
    marker.color = ColorRGBA(r, g, b, a)

    return marker

if __name__ == '__main__':
    # node init
    rospy.init_node('marker_dae_test')
    rate = rospy.Rate(1)
    # publishers
    pub = rospy.Publisher('markers', MarkerArray, queue_size=1)
    # --------------------------------------------------------- Markers -
    msg = MarkerArray()
    # cube
    cube = createMarker('world', Marker.MESH_RESOURCE, 1)
    cube.mesh_resource = "package://env_3d/data/nil_link_mesh.dae"
    cube.mesh_use_embedded_materials = True
    # add markers to list
    msg.markers.append(cube)
    while not rospy.is_shutdown():
        pub.publish(msg)
        rate.sleep()
```
And here is the [.dae file](https://github.com/garaemon/rviz_collada_marker/blob/master/nil_link_mesh.dae).

Am I missing something ?

exo-core, 2018-01-29:
I just ran into the same issue.

This is how my mesh is supposed to look like (screenshot from Blender):
[image]

And this is how it appears in RVIZ, build 1.12.15-0xenial-20180105-172609-0800 amd64:
[image]

Code of my mesh:
```
<?xml version="1.0" encoding="utf-8"?>
<COLLADA xmlns="http://www.collada.org/2005/11/COLLADASchema" version="1.4.1">
  <asset>
    <contributor>
      <author>Blender User</author>
      <authoring_tool>Blender 2.79.0</authoring_tool>
    </contributor>
    <created>2018-01-29T13:36:13</created>
    <modified>2018-01-29T13:36:13</modified>
    <unit name="meter" meter="1"/>
    <up_axis>Z_UP</up_axis>
  </asset>
  <library_images/>
  <library_effects>
    <effect id="red-effect">
      <profile_COMMON>
        <technique sid="common">
          <phong>
            <emission>
              <color sid="emission">0 0 0 1</color>
            </emission>
            <ambient>
              <color sid="ambient">0 0 0 1</color>
            </ambient>
            <diffuse>
              <color sid="diffuse">1 0 0 1</color>
            </diffuse>
            <specular>
              <color sid="specular">0.25 0.25 0.25 1</color>
            </specular>
            <shininess>
              <float sid="shininess">50</float>
            </shininess>
            <index_of_refraction>
              <float sid="index_of_refraction">1</float>
            </index_of_refraction>
          </phong>
        </technique>
      </profile_COMMON>
    </effect>
    <effect id="green-effect">
      <profile_COMMON>
        <technique sid="common">
          <phong>
            <emission>
              <color sid="emission">0 0 0 1</color>
            </emission>
            <ambient>
              <color sid="ambient">0 0 0 1</color>
            </ambient>
            <diffuse>
              <color sid="diffuse">0 0.8 0 1</color>
            </diffuse>
            <specular>
              <color sid="specular">0.25 0.25 0.25 1</color>
            </specular>
            <shininess>
              <float sid="shininess">50</float>
            </shininess>
            <index_of_refraction>
[... 85 lines cut ...]
      </mesh>
    </geometry>
  </library_geometries>
  <library_controllers/>
  <library_visual_scenes>
    <visual_scene id="Scene" name="Scene">
      <node id="Cylinder" name="Cylinder" type="NODE">
        <matrix sid="transform">1 0 0 0 0 1 0 0 0 0 1 0 0 0 0 1</matrix>
        <instance_geometry url="#Cylinder-mesh" name="Cylinder">
          <bind_material>
            <technique_common>
              <instance_material symbol="red-material" target="#red-material"/>
              <instance_material symbol="green-material" target="#green-material"/>
              <instance_material symbol="blue-material" target="#blue-material"/>
            </technique_common>
          </bind_material>
        </instance_geometry>
      </node>
    </visual_scene>
  </library_visual_scenes>
  <scene>
    <instance_visual_scene url="#Scene"/>
  </scene>
</COLLADA>
```

ejalaa12, 2018-01-29:
@exo-core I also noticed that if you use image textures in Blender then Rviz displays it correctly. But everything else that uses classic colored materials doesn't work. (it can be a workaround while waiting for a correct fix)

exo-core, 2018-01-29:
@ejalaa12 I also noticed. The tetured meshes from the pr2-description package are working fine.

I also have a feeling that there is something else wrong with mesh format markers. Since I am using them instead of simple arrows for my interactive markers RVIZ crashes from time to time, while I am moving them. Unfortunately I did not manage to find a reproducable situation to closer investigate the issue - maybe it's also my code that's messed up... GDB backtrace points somewhere to Ogre drawing routines though...

wjwwood, 2018-01-30:
Either @dhood or I will try to reproduce what you've described as soon as we can.

willcbaker, 2018-02-22:
+1, have white meshes when textured using basic material in blender (not UV map)

wjwwood, 2018-06-11:
I think this was fixed a while back by https://github.com/ros-visualization/rviz/pull/1132 (it claims to do so), please comment if this is not the case for someone.

wjwwood, 2018-06-11:
Ah, I already closed/reopened this... Sorry.

willcbaker, 2018-07-06:
@wjwwood confused by the closed/open, this is still an issue, right? I still have white meshes in kinetic building from src and the deb

wjwwood, 2018-07-09:
Still an issue.

ana-GT, 2018-08-10:
Still an issue in ROS kinetic (white meshes for .dae meshes)

Master-Matt, 2018-08-30:
Trying to view this in in Rviz shows only a white box.
I'm using Kinetic, so Gazebo 7 and Rviz 1.12.16
[april_box.zip](https://github.com/ros-visualization/rviz/files/2337458/april_box.zip)

ejalaa12, 2019-05-06:
I've decided to have a look to understand the full issue where the problem comes from by looking at every possible scenario.

## use_embedded_materials=False (works fine)
When `use_embedded_materials=False`, the material (texture or classical color material) is replaced by whatever color is defined in the published `Marker` msg

## use_embedded_materials=True (issue)
### r,g,b,a = 0,0,0,0
Textured materials are used correctly. But `'classical color' materials are white.

### r,g,b = 0,0,0 ; alpha !=0
Textured materials *and* classical materials are black with **alpha** transparency

### r,g,b != 0,0,0; alpha = 0
Everything is invisible (because alpha=0)

#### r,g,b,a != 0,0,0,0
Textured materials are tinted with r,g,b and have the alpha transparency
Classical materials are replaced with r,g,b, and have the alpha transparency

---
It's weird that this issue keeps coming back, it has been solved here apparently https://github.com/ros-visualization/rviz/pull/752 by creating passes but his solution has disappeared since then from the source code. I tried to write it back, but my knowledge of Ogre is very limited, and I can't figure it out.

I've put the files for testing this on a gist that you can find [here](https://gist.github.com/ejalaa12/dc5edb5d33f13781141fb6d75224afed)

xbroquer, 2019-07-18:
Reproduced on Melodic, I re-integrated #752 on top of melodic branch

https://github.com/xbroquer/rviz/commit/aa6f232e61c8a42ac69ecbb3531d472a8e141379

rhaschke, 2019-09-09:
I filed a new solution for this issue: #1424. Please verify that this solution is satisfying to you.

PR #1132 (2017-08-04): Fix for 1120
Fixes #1120.

Basically #1084 caused a regression that caused markers which had a mesh text and color to render white until the color or alpha was updated, after which point it would render correctly.

I also improved the marker mesh test executable to test a few of these things. Here's a screen shot of the behavior with this pr (the correct behavior imo):

[image]
PR #1424 (2019-09-20): Fix tinting of mesh markers
Fix #1120. This issue has a long story:
* Originally, it was fixed in #752 by adding a new rendering pass to realize color tinting.
* #893 essentially reverted this patch. The motivation at that time was that the second pass has essentially overwritten the first pass when the alpha value of the message color was `1.0`. Also, it wasn't possible to make the mesh transparent by setting a smaller alpha value.
* #1084 attempted to fix the white rendering issue again, but actually only skipped color tinting at all, when the message arrived the first time.
* This caused ju [... cut]
closed 2019-09-20 by rhaschke