ros-visualization/rviz #1126: Regular rviz startup crash on indigo

broesdecat, 2017-07-20:
Hi,

Quite often (as in, half the time), my rviz crashing while loading, with the stack below.
Once started, it typically continues to work fine.

Any known solutions or workarounds? It resembles what is explained in http://wiki.ros.org/rviz/Troubleshooting ("Segfault in i915_dri.so") but for a  more recent lib.

Some more details: 
- running indigo on ubuntu 14.04 within docker
- mostly visualising the kuka iiwa robotarm, without additional scenery

Stack:
#0  0x00007fc3fea23168 in ?? () from /usr/lib/x86_64-linux-gnu/dri/r600_dri.so
#1  0x00007fc3fea223d1 in ?? () from /usr/lib/x86_64-linux-gnu/dri/r600_dri.so
#2  0x00007fc3fea28dff in ?? () from /usr/lib/x86_64-linux-gnu/dri/r600_dri.so
#3  0x00007fc3fa6fd6fb in Ogre::GLRenderSystem::clearFrameBuffer(unsigned int, Ogre::ColourValue const&, float, unsigned short) () from /usr/lib/x86_64-linux-gnu/OGRE-1.8.0/RenderSystem_GL.so
#4  0x00007fc422291b1d in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.8.1
#5  0x00007fc422106831 in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.8.1
#6  0x00007fc4222560de in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.8.1
...
#12 0x00007fc4222780c0 in Ogre::Root::renderOneFrame() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.8.1
#13 0x00007fc426318395 in rviz::VisualizationManager::onUpdate() () from /opt/ros/indigo/lib/librviz.so
#14 0x00007fc42169c87a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
...
QCoreApplication::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#27 0x00007fc426314bbe in rviz::VisualizationFrame::initialize(QString const&) () from /opt/ros/indigo/lib/librviz.so
#28 0x00007fc42631cd9b in rviz::VisualizerApp::init(int, char**) () from /opt/ros/indigo/lib/librviz.so
#29 0x0000000000400c2d in main ()

Thanks,
Broes

wjwwood, 2017-07-24:
As you pointed out, it appears to have the problem somewhere within the graphics card driver. Without that specific machine configuration I'm likely unable to help you with that.

Also, can you say if it does or does not have this issue outside of docker? Perhaps if you search for issues online about your card in combination with docker that might turn something up. Also, does something like this happen if you replace rviz with glxgears or some other opengl program?

broesdecat, 2017-07-25:
I investigated a bit further. Two conclusions: 
- The issue is not docker related, as rviz also crashes (although apparently less often) on the host OS.
- Gazebo seems to have [a simular issue](https://bitbucket.org/osrf/gazebo/issues/2304/gzclient-segfault-on-startup-using-amd).

Some machine details:
- host OS is Mint 18.1 (aka Ubuntu 16.04), with a Radeon HD 6970M graphics card
- docker OS is ros:indigo (aka Ubuntu 14.04).
Glxgears seems to be running fine, both inside the docker container and in the hosted OS.

As Canonical/ATI discontinued the catalyst driver support on 16.04, I have no idea on an easy workaround.

dhood, 2017-12-18:
@broesdecat if this is still an issue for you can you try the fix proposed in the gazebo thread by changing `render_viewport->setClearEveryFrame(true);` to `false` and let us know if that makes a difference?
https://github.com/ros-visualization/rviz/blob/7970ba08cee3810cfa1609c3b0f5136970eb2f7c/src/rviz/selection/selection_manager.cpp#L668