ros-visualization/rviz #1135: MarkerDisplay limits you to 100 Markers

mogumbo, 2017-08-07:
As far as I can tell, you should be able to have more than 100 Markers in your scene by doing Add->Marker in RViz and increasing the Queue Size above 100. However, this appears to have no effect. I can only see 100 Markers in RViz. What does work is to hardcode a new default size for queue_size_property_ in the MarkerDisplay constructor.

wjwwood, 2017-08-07:
If you're sending many markers at once (only time the queue size is important) then you should probably use a MarkerArray. Also when you want to scale up things past hundreds you should try to use a "list" style marker rather than lots of individual markers if you are able, see:

http://wiki.ros.org/rviz/DisplayTypes/Marker#Rendering_Complexity_Notes

Either way, the property should work as you've described (it doesn't make sense that hard coding a new value works but changing the property does not).

mogumbo, 2017-08-07:
Thank you for the tips. Trying MarkerArray was on my todo list as a possible solution, but I'm not familiar with "list" style markers yet. I'll investigate those too.

PR #1429 (2019-09-28): Adapt subscriber size with TF queue size
Fixes #1135. Obviously, when increasing the TF filter's queue size, we also need to adapt the subscriber's queue size, i.e. re-subscribe. Otherwise, we cannot even receive enough messages.
closed 2019-09-28 by rhaschke