ros-visualization/rviz #1136: Covariance visualization spams ROS_WARN

agoddijn, 2017-08-08:
The covariance visualization spams ROS_WARN when the eigenvalues of the covariance matrix are negative. It tries to find the root of the eigenvalue which is NaN and complains. I'm not sure about the math behind the covariance values but would it be possible to take the absolute value of the eigenvalues, or maybe use singular values instead? Or maybe just not print the warning and deal with NaNs?

dhood, 2017-08-10:
For covariance matrices the calculated eigenvalues should always be positive, and if not there is an issue in the covariance matrix and it's appropriate to warn users. If you are seeing the warning you might want to check that the matrix satisfies the properties of a covariance matrix (e.g. that it's symmetric).
If you have a specific example where a valid covariance matrix isn't able to be displayed could you please provide a code snippet to reproduce the issue?

dhood, 2017-08-22:
@agoddijn I'll close this for now since it doesn't seem there's any work to be done on our side, but feel free to re-open if you have an example that makes you think otherwise

closed 2017-08-22 by dhood

dhood, 2017-08-22:
correction: please comment here and we'll re-open on your behalf