ros-visualization/rviz #1137: RViz dies when receiving an incomplete Odometry message

sharst, 2017-08-09:
Steps to reproduce: 
* Add odometry display
* Point to a topic
* Publish an odometry message with a (0x, 0y, 0z, 0w)-quaternion on this topic.

Rviz dies with the following traceback:
[ INFO] [1502309518.274159660]: rviz version 1.12.10
[ INFO] [1502309518.274207080]: compiled against Qt version 5.5.1
[ INFO] [1502309518.274219210]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1502309518.411463334]: Stereo is NOT SUPPORTED
[ INFO] [1502309518.411575145]: OpenGl version: 3 (GLSL 1.3).
rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
Aborted (core dumped)

It would be nicer if this error were catched and a warning pointing to the origin of this problem be presented to the user.

wjwwood, 2017-08-09:
That would be great. I don't have time to do that right now, but I'm always happy to take a pull request.

skohlbr, 2017-08-10:
Pretty sure this sensitivity to non-normalized (in most cases probably all-zero) quaternions appears  also in other Displays used in rviz. I know I saw a OgreAxisAlignedBox segfault when we rviz received a "bad" quaternion in some other message (cannot remember which one unfortunately).

dhood, 2017-08-17:
thanks for pointing that out @skohlbr. @sharst has fixed this particular crash in https://github.com/ros-visualization/rviz/pull/1139 so I'll close this issue; I've opened https://github.com/ros-visualization/rviz/issues/1142 to track that there are other parts in the codebase that could use this improvement (contributions welcome!)

closed 2017-08-17 by dhood