ros-visualization/rviz #1138: Change RViz tool display name and add multiple instances of same tool

jay3ss, 2017-08-10:
I can added multiple instances of the same tool to the tool bar, but I can't change the name of each instance of the tool. Currently this isn't possible because the names are hardcoded and set via the `setName` function. 

It would also be great if multiple instances of the same tool such as `2D Nav Goal` and `2D Pose Estimate` could be added via the `+` button instead of having to add them through a custom configuration file.

This is in response to [this question](https://answers.ros.org/question/268445/change-rviz-tool-display-name/) on ROS Answers.

wjwwood, 2017-08-10:
That makes sense, but I probably won't have time to implement it right now. If you want to implement it and open a pull request though, I'll try to get it merged in.

jay3ss, 2017-08-11:
Sure, I'll take a stab at it. I can't find guidelines on contributing, are there any?

wjwwood, 2017-08-17:
@jay3ss we don't have any written down guidelines, but generally speaking just try to maintain the code style in the file you modify and open a pr against `kinetic-devel`.

jay3ss, 2017-08-17:
@wjwwood will do, thank you.

rhaschke, 2019-09-24:
Closing due to inactivity.

closed 2019-09-24 by rhaschke