ros-visualization/rviz #1142: Check for normalized quaternion before passing to Ogre

dhood, 2017-08-17:
As reported in https://github.com/ros-visualization/rviz/issues/1137, crashes occur when non-normalized quaternions are passed through to Ogre.

A fix for one instance was implemented in https://github.com/ros-visualization/rviz/pull/1139 but there are other parts in the codebase that are susceptible. Ideally the logic added in that PR would be factored out and reused in other places (as is done with [validate_floats.h](https://github.com/ros-visualization/rviz/blob/7970ba08cee3810cfa1609c3b0f5136970eb2f7c/src/rviz/validate_floats.h)).

This would be a good task for someone new to the codebase if anyone is interested

StefanFabian, 2017-10-18:
I've implemented this but I have a question regarding the license in the new validate_quaternions.h.
Should I keep "Willow Garage, Inc" or replace it with my name?
Apart from their existence, I don't really know much about licenses.

dhood, 2017-12-12:
closed by https://github.com/ros-visualization/rviz/pull/1167

closed 2017-12-12 by dhood