ros-visualization/rviz #1147: Multiple licenses?

MasterEric, 2017-08-30:
RVIZ is stated, both in `package.xml` and on the [ROS wiki](http://wiki.ros.org/rviz), to be licensed under both the BSD and the Creative Commons license. Is this correct? Shouldn't an application specify only a single license?

wjwwood, 2017-08-30:
The wiki content is based on the `package.xml`. I just did a quick search, but I think the creative commons only applies to the documentation in this package, and not the Code.

The oldest blame on that line goes to @hersh, can you comment?

jjarquin, 2017-09-28:
I would also like this to be clarified.

wjwwood, 2017-09-28:
All I can do is search and ask for older developers to comment.

Based on my search there is no code licensed under the creative commons and I believe quite strongly it would have only applied to the documentation in the package (not code documentation, but like main page doxygen type stuff).

I'll ping some others too, maybe they will have some confirmation @hershwg @hersh @dgossow.

----

I'll close this with the working idea that the code is BSD and only documentation was creative commons until some evidence comes up to the contrary.

----

Also to answer this question...

> Shouldn't an application specify only a single license?

No, an application can use many pieces of software of various licenses.

closed 2017-09-28 by wjwwood

dgossow, 2017-09-28:
Maybe CC was meant to apply to the icons and other graphics?

wjwwood, 2017-09-28:
> Maybe CC was meant to apply to the icons and other graphics?

That sounds like it might be the case. Just looking at a few of the `.svg` they do mention CC, but I don't know if that's just the default for inkscape or if that was on purpose, e.g.:

https://raw.githubusercontent.com/ros-visualization/rviz/kinetic-devel/icons/crosshair.svg

At any rate, I feel confident that none of the code is CC.

dgossow, 2017-09-29:
I made those SVG files. I don't remember choosing a license, but it seems appropriate. I don't know if you can apply the BSP license to graphics.

wjwwood, 2017-10-02:
@dgossow it would help us out if you clarified a license (as the original author) for the images and/or icons you created, maybe as a pr adding a readme to the `icons`, `image_src`, and `images` folders or modifying the top-level readme. Specifically which version of the CC license you'd like to use.

Oddly, I can't find much information on whether or not images and icons can be licensed under the BSD license we are using. I would guess yes, but I don't know.

dgossow, 2017-10-02:
https://github.com/ros-visualization/rviz/pull/1155

wjwwood, 2017-10-02:
Thanks @dgossow, much appreciated.