ros-visualization/rviz #1148: Allow to use scroll buttons in QTabWidgets created by RViz

VictorLamoine, 2017-09-11:
If you stack a lot of widgets inside RViz, the labels becomes hard to read like this:
[image]

Solutions:
- Use shorter tab names (+ a tooltip)
- Expand the tab so that labels becomes easy to read again:
[image]
- Allow to use [scroll buttons](http://doc.qt.io/qt-5/qtabwidget.html#usesScrollButtons-prop) in the QTabWidget:
  - [image] (without scroll buttons)
  - [image] (with scroll buttons)

Last solution is by far the best in my opinion.
I think one solution is to subclass the tab bar and add a right click action to choose between scroll buttons, no scroll buttons.

VictorLamoine, 2017-10-23:
I can't find the code that manages that part. I searched for QTabWidget but it's only returning one results which is not what I'm looking for (`src/rviz/add_display_dialog.cpp`).

Which part of the RViz code manages the stacking/moving of whole RViz panels?

wjwwood, 2017-10-26:
@VictorLamoine I believe it is just done by `QtMainWindow` (of which `VisualizationFrame` inherits). Maybe what you want is https://doc.qt.io/qt-5/qmainwindow.html#DockOption-enum?

VictorLamoine, 2017-10-26:
Thanks @wjwwood, it helped me a lot because I did not know the "dock widget" names.

I made a test and this is exactly what I wanted:

# Default behaviour
[image]

# Don't elide text in QTabBar
[image]

Now I need to find a place/signal to put this code (each time a panel is added or moved); the code itself is very straightforward:

https://github.com/ros-visualization/rviz/compare/kinetic-devel...VictorLamoine:QTabWidget_elide?expand=1

gavanderhoorn, 2017-10-26:
Diff without whitespace changes: https://github.com/ros-visualization/rviz/compare/kinetic-devel...VictorLamoine:QTabWidget_elide?expand=1&w=1.

PR #1168 (2018-04-26): Disable dock widget text eliding
Fixes #1148 

Diff without trailing whitespace changes:
https://github.com/ros-visualization/rviz/compare/kinetic-devel...VictorLamoine:QTabWidget_elide?w=1

The new slot `onDockPanelChange` checks for every QTabBar there is in the main window and makes sure text is not elided inside it.

This slot is triggered
- When a config file is loaded
- When a panel is created
- When a panel is moved

# Before
[image]

# After
[image]
closed 2018-04-26 by dhood