ros-visualization/rviz #1151: rviz doesn't show any shape

Maik93, 2017-09-23:
Hi,
I'm trying to build my first URDF model, following this tutorial:
http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch

Booth replicating what's done there, and cloning [this repo](https://github.com/ros/urdf_tutorial) too, I can't see any mash or shape in rviz, only transformations; here's what I see from [chap.4](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch#Material_Girl) code:

[image]

There should be cylinders and squares...

Anyone experiencing this issue? I'm running ROS Lunar, Ubuntu 17.04 x64 on VirtualBox 5.1.18 (with "3D acceleration" disabled because i don't know why, if it's enabled rviz trows segmentation faults at start)

wjwwood, 2017-09-25:
Please post the output in the terminal where you run rviz. The most likely case is that there is an issue with locating the meshes or processing the urdf. However, that usually results in an error in the status of the `RobotModel` display.

ycantin, 2017-10-07:
Hi,
Same thing here : Debian stretch amd64, nvidia with nouveau driver, ROS Lunar.
Rviz don't display any urdf's basic geometry (but dae meshes are ok).

The ogre meshes are loaded : i've tested with older ones and seen the warning.

I've tweaked  display.launch from urdf_tutorial to add the -v -l --log-level-debug args to rviz, 
here's the log :

[roslaunch_stdout.log](https://github.com/ros-visualization/rviz/files/1366360/roslaunch_stdout.log)
[rviz-4-stdout.log](https://github.com/ros-visualization/rviz/files/1366359/rviz-4-stdout.log)

JanGunheD, 2017-10-19:
Same problem here, can be verified using
cd /opt/ros/lunar/share/urdf_tutorial/
roslaunch urdf_tutorial display.launch model:=urdf/06-flexible.urdf

[roslaunch-Emulator-1-Ubuntu-23284.log](https://github.com/ros-visualization/rviz/files/1399289/roslaunch-Emulator-1-Ubuntu-23284.log)
[rviz-4-stdout.log](https://github.com/ros-visualization/rviz/files/1399290/rviz-4-stdout.log)
[robot_state_publisher-3-stdout.log](https://github.com/ros-visualization/rviz/files/1399291/robot_state_publisher-3-stdout.log)
[master.log](https://github.com/ros-visualization/rviz/files/1399292/master.log)
[rosout.log](https://github.com/ros-visualization/rviz/files/1399293/rosout.log)
[joint_state_publisher-2.log](https://github.com/ros-visualization/rviz/files/1399294/joint_state_publisher-2.log)
[rosout-1-stdout.log](https://github.com/ros-visualization/rviz/files/1399295/rosout-1-stdout.log)

JanGunheD, 2017-10-23:
This is the terminal output of the tutorial example:

[TerminalOutput.txt](https://github.com/ros-visualization/rviz/files/1408018/TerminalOutput.txt)

(Commands:
cd /opt/ros/lunar/share/urdf_tutorial/
roslaunch urdf_tutorial display.launch model:=urdf/06-flexible.urdf
)

JanGunheD, 2017-10-27:
I just found out by accident, that primitive spheres are displayed, if they have a radius of >= 1, if they are smaller, they are not displayed,

Maik93, 2017-11-05:
Hi,
just realized that this problem appears only on VirtualBox, while using Ubuntu booted from partition shows correctly all shapes. Thus i believe that OpenGL2 (used in VirtualBox) is the cause of our problem.

JanGunheD, 2017-11-05:
For me it happens on 3 PCs using Ubuntu 17.04 natively without a VirtualBox.

Maik93, 2017-11-05:
ook, in that case, the only other difference with my machines is that I've got Ubuntu 17.10 on VirtualBox and 16.04LTS on laptop partition.

dbasa, 2017-11-16:
Same thing here,

I'm running ROS Lunar, Ubuntu 17.04 x64 natively (no virtual box). dae meshes are shown, but primitives are not visualized.

I also have some strange issues with joint positions and joint orientation. They are somehow incorrect. orientation is strangely skewed, as if conversion between rotation matrix and quaternions is buggy. Can somebody confirm this? Should I open a new issue?

dbasa, 2017-11-20:
As reported I had the same problem, but furthermore a problem with joint position and orientation. I could track this down to an error in `urdfdom_headers` package. Once I fixed this the geometries do show up. In my case it was a problem with `locales`. `urdfdom_headers` uses `std:stod` to parse floating point values from `urdf` files. `std:stod` is depending on your locale.
In that light, what @JanGunheD posted makes absolutely sense. This error cuts all values below one to zero.

Could somebody check if changing the locale `LC_NUMERIC` to e.g. `en_US.UTF-8` fixes this error, as it does on my end?
You could run `export LC_NUMERIC="en_US.UTF-8` for testing.

JanGunheD, 2017-11-20:
Yes,
export LC_NUMERIC="en_US.UTF-8"
solved the problem!

closed 2019-09-24 by rhaschke