ros-visualization/rviz #1152: Arial font exception

davetcoleman, 2017-09-27:
Why I would still be getting:

```
OGRE EXCEPTION(5:): Could not find font Arial in MovableText::setFontName
```

after 1) updating to the latest 16.04 debian and 2) even after building Rviz from source with [this PR](https://github.com/ros-visualization/rviz/pull/1141)?

wjwwood, 2017-09-27:
You're mixing versions I guess. Can you post the `ogre.log` file?

mmoerdijk, 2017-09-27:
This has to do with the widgets provided by moveit, there the Arial font is still used.

wjwwood, 2017-09-27:
@mmoerdijk I see, that makes sense (would have seen that in the `ogre.log` probably).

Sorry for the disruption, but we had to exchange this font for licensing reasons, so it was unavoidable.

The fix should be as easy as updating the moveit plugin to replace any uses of `Arial` with `Liberation Sans`.

wjwwood, 2017-09-27:
I'll close this for now.

closed 2017-09-27 by wjwwood

gavanderhoorn, 2017-09-28:
@wjwwood: I noticed the PR that removed Arial from RViz, but a lot of devs/users probably have not. With the customisability that RViz supports through its plugins, it would probably be nice to give #1141 a bit more visibility to save people some frustration?

MoveIt is potentially just one plugin that breaks because of this.

wjwwood, 2017-09-28:
> Could it be that fonts-liberation is not installed on those (Ubuntu) machines?

"Liberation Sans" is distributed with rviz, see: https://github.com/ros-visualization/rviz/tree/kinetic-devel/ogre_media/fonts

> I still think some more visibility (Discourse post?) would be good.

I'll send a message.

gavanderhoorn, 2017-09-28:
> > Could it be that fonts-liberation is not installed on those (Ubuntu) machines?
> 
> "Liberation Sans" is distributed with rviz, see: https://github.com/ros-visualization/rviz/tree/kinetic-devel/ogre_media/fonts

the OP of the ROS Answers question implies that after installing that pkg things started working. That is unexpected then.

wjwwood, 2017-09-28:
It is, because Ogre needs to have the font on the ogre resource path and you need the Ogre specific "fontdef" as well I think:

https://github.com/ros-visualization/rviz/blob/kinetic-devel/ogre_media/fonts/liberation_sans.fontdef

Also the moveit code is still asking for Arial, not Liberation Sans, so install Arial would fix the problem, not installing liberation sans...

wjwwood, 2017-09-28:
https://discourse.ros.org/t/notice-removal-of-arial-in-rviz-caused-breaking-changes/2791

emila14, 2017-10-11:
Hi, how do I update the moveit plugin to replace any uses of `Arial` with` Liberation Sans`?