ros-visualization/rviz #1154: Pixmap error

BrunoXL, 2017-09-28:
I am getting a `could not load pixmap` error on launching rviz from command line.

[image]

wjwwood, 2017-09-28:
Please do not post a picture of text. I will close this until it is updated, at which I'll reopen it.

Thanks.

wjwwood, 2017-09-28:
Also, please include the output of `env | grep ROS` and the `ogre.log` file if you can (see `rviz -h` for info on how to get the `ogre.log` file).

BrunoXL, 2017-10-01:
@wjwwood I have updated this issue as you asked me. Could you reopen it?

wjwwood, 2017-10-02:
Do those files exist, for example does `/opt/ros/kinetic/share/rviz/icons/rotate.svg` exist?

BrunoXL, 2017-10-02:
Yes, I checked some of them and they really exist.

wjwwood, 2017-10-02:
Please link the `ogre.log` file, again see `rviz -h` for how to generate it.

BrunoXL, 2017-10-04:
For some reason running `rosrun rviz rviz -l` is not generating the `ogre.log` file. Therefore, I redirect the command line output to the I`m linking.

[ogre.log](https://github.com/ros-visualization/rviz/files/1357398/ogre.log)

wjwwood, 2017-10-04:
@BrunoXL It does generate it, but it's probably in `~/.ros`, since that is likely the working directory of `rosrun`.

wjwwood, 2017-10-04:
`rviz -l` might "just work" and put the file in the current directory.

wjwwood, 2017-10-05:
How do you have your `ogre` installed? It looks like it cannot load any svg files, which I think could only really happen if ogre was built without support for it (not even sure that's possible).

Please try to provide the `ogre.log` file produced by `rviz -l` based on my previous two comments.

BrunoXL, 2017-10-05:
I do have ogre installed.

    ` bruno@glei:~$ sudo dpkg -s libogre-1.9.0v5   
      Package: libogre-1.9.0v5
      Status: install ok installed
      Priority: optional
      Section: libs
      Installed-Size: 11160 
      Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>`

The only log files generated are those from `ros`.

Neoplanetz, 2017-10-17:
I have same problem with BrunoXL. Did you solve this isue??

BrunoXL, 2017-10-17:
@Neoplanetz I could not solve it so far. If you run  `rviz -l` a log file is generated?

wjwwood, 2017-10-17:
@BrunoXL what's in `/home/bruno/rodain/src`?

wjwwood, 2017-10-17:
Or even, what does `rospack find rviz` return?

BrunoXL, 2017-10-17:
`/opt/ros/kinetic/share/rviz`

wjwwood, 2017-10-17:
@BrunoXL I cannot reproduce your issue on any of my setups.

The code which is producing this error is here:

https://github.com/ros-visualization/rviz/blob/7970ba08cee3810cfa1609c3b0f5136970eb2f7c/src/rviz/load_resource.cpp#L85

It is based solely on the result of `QPixmap::load()`:

http://doc.qt.io/qt-5/qpixmap.html#load

I don't know what else to do to help you. Perhaps the file cannot be read for some reason, permissions or corruption, I don't know. You could try writing a simple Qt program that tries to load those files to further debug it.

I'm sorry I could not figure out the issue for you.

mescarra, 2018-01-19:
I'm having the exact same issue here. Files are actually there but the error appears all the same. BTW, I can realize it's not loading the images because the default_class_icon.svg appears instead of those images.

This is one of those things you can live with, maybe that's why this thread is dead.

francescomilano172, 2019-05-27:
I had the same problem, due to a corrupted Qt5 installation. You could try setting the following environmental variable: `export QT_DEBUG_PLUGINS=1` and then running `rviz` in the same terminal. In my case the debug prints said: `Cannot load library /usr/lib/x86_64-linux-gnu/qt5/plugins/imageformats/libqsvg.so: (libQt5Svg.so.5: cannot open shared object file: No such file or directory)`. So I solved the issue by reinstalling the related Qt plugin (`sudo apt-get --reinstall install libqt5svg5`).

pirobot, 2019-10-02:
Make sure you don't have another Qt library in your LD_LIBRARY_PATH ahead of what ROS needs.  I had the same problem because I was also trying VREP and had the following lines in my ~/.bashrc file:

jacobhuesman, 2019-10-27:
Noticed this bug while running rviz2 from a docker container. It looks like the debian package for rviz2 doesn't include `libqt5svg5` as a dependency. Installing the package manually fixed the issue for me.

davetcoleman, 2021-08-31:
In ROS2 Rolling I'm getting a lot of errors regarding this icons, and @jacobhuesman's fix did not work. I'm seeing stuff like:

> [moveit_studio_ui-1] [rviz_common:debug] Error retrieving file [file:///opt/ros/rolling/share/rviz_default_plugins/icons/classes/Intensity.svg]: Couldn't open file /opt/ros/rolling/share/rviz_default_plugins/icons/classes/Intensity.svg, at /tmp/binarydeb/ros-rolling-rviz-common-8.7.0/src/rviz_common/load_resource.cpp:55

Checking the file system, the file does not exist. It appears in ROS 2 some of the icons where added in this PR https://github.com/ros2/rviz/pull/235/files but not the ones I'm missing, including ``Intensity.svg``

closed 2021-08-31 by rhaschke

jasonbeach, 2022-10-18:
> Noticed this bug while running rviz2 from a docker container. It looks like the debian package for rviz2 doesn't include libqt5svg5 as a dependency. Installing the package manually fixed the issue for me.

@jacobhuesman's solution of installing `libqt5svg5` worked for me.