ros-visualization/rviz #1159: RViz crashes (segmentation fault) on launch

diogoalmeida, 2017-10-05:
Hi,

My installation of RViz started segfaulting everytime I try to run it. It was working until recently, but I had the unfortunate idea of doing an apt-get upgrade and since then I haven't been able to run RViz. I have tried all the solutions proposed in the Troubleshooting guide, and I've searched both in the issues page and the ROS answers website, but haven't found a solution. I run ubuntu 16.04 and ROS kinetic. When I run RViz, this is the output:

```
$ rosrun rviz rviz         
[ INFO] [1507195009.868024274]: rviz version 1.12.13
[ INFO] [1507195009.868098854]: compiled against Qt version 5.5.1
[ INFO] [1507195009.868116149]: compiled against OGRE version 1.9.0 (Ghadamon)
[1]    2980 segmentation fault (core dumped)  rosrun rviz rviz
```

I've made a [gist](https://gist.github.com/diogoalmeida/1ad00cd7978da80591c93d934fdc8d99) with the gdb backtrace, ogre log and system info, and I've uploaded the core file to [google drive](https://drive.google.com/open?id=0B_gHe0UTh5trZG96Q3VjTmNZVVk). Hopefully this will be helpful.

I believe this to be a driver issue, but I have no idea what I could do. I would like to avoid reinstalling ubuntu. Thank you very much!

diogoalmeida, 2017-10-10:
If I remove the intel video drivers (`sudo apt-get remove xserver-xorg-video-intel`) then RViz does not crash. However, the graphics performance goes down substancially :). Am I missing something here?

dpipemazo, 2017-10-23:
Am seeing the same issue. GDB callstack below
```
Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007fffa7b8bc09 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#2  0x00007fffa7b5627f in Ogre::GLStateCacheManager::switchContext(long) ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#3  0x00007fffa7b4704d in Ogre::GLRenderSystem::initialiseContext(Ogre::RenderWindow*) ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#4  0x00007fffa7b4fb13 in Ogre::GLRenderSystem::_createRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*) ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007ffff3d3588c in Ogre::Root::createRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff7ad74da in rviz::RenderSystem::tryMakeRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, s---Type <return> to continue, or q <return> to quit---
td::allocator<char> > const&, unsigned int, unsigned int, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*, int) ()
   from /opt/ros/kinetic/lib/librviz.so
#7  0x00007ffff7ad8c9e in rviz::RenderSystem::makeRenderWindow(unsigned long, unsigned int, unsigned int, double) ()
   from /opt/ros/kinetic/lib/librviz.so
#8  0x00007ffff7adb175 in rviz::RenderSystem::RenderSystem() () from /opt/ros/kinetic/lib/librviz.so
#9  0x00007ffff7adb2d5 in rviz::RenderSystem::get() () from /opt/ros/kinetic/lib/librviz.so
#10 0x00007ffff7ad6691 in rviz::QtOgreRenderWindow::QtOgreRenderWindow(QWidget*) () from /opt/ros/kinetic/lib/librviz.so
#11 0x00007ffff7afcd19 in rviz::RenderPanel::RenderPanel(QWidget*) () from /opt/ros/kinetic/lib/librviz.so
#12 0x00007ffff7b47bc0 in rviz::VisualizationFrame::initialize(QString const&) () from /opt/ros/kinetic/lib/librviz.so
#13 0x00007ffff7b52e29 in rviz::VisualizerApp::init(int, char**) () from /opt/ros/kinetic/lib/librviz.so
#14 0x0000000000400c90 in main ()
```

wjwwood, 2017-10-23:
Looks like the segfault is within Ogre. Based on my experience I would guess (but I don't know for sure) that Ogre is assuming some GLX function is available, but it is not provided by your VM's video driver and therefore Ogre trys to call a function pointer that is `nullptr`. Without being able to reproduce it here, I doubt I can make progress on this issue. Even if I could, I fear all I could do is improve the error handling so that it throws an error that a required feature is unsupported (in fact I did this for Ogre 1.10: https://github.com/OGRECave/ogre/pull/507), avoiding the segfault but not making is so rviz can run.

dpipemazo, 2017-10-23:
Thanks for the super-fast response. I agree on your analysis; it seems to be working better in Parallels for Mac for what it's worth if anyone else is running into the issue in VirtualBox.

diogoalmeida, 2017-10-24:
I can add more info: I am running a laptop which docks with a displaylink adapter. If I configure my system to not shut down the laptop monitor when docked, then RViz starts, otherwise I get the segfault.

VictorLamoine, 2018-11-19:
Same problem here:
- Xubuntu 18.04
- Intel Corporation HD Graphics 530

```bash
#0  0x00007fff7f233496 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#1  0x00007fff7f237ee0 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#2  0x00007fff7f25c933 in Ogre::GLRenderSystem::GLRenderSystem() () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#3  0x00007fff7f2576e4 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#4  0x00007ffff31ada26 in Ogre::Root::installPlugin(Ogre::Plugin*) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#5  0x00007fff7f24079a in dllStartPlugin () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#6  0x00007ffff31adc61 in Ogre::Root::loadPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff7ad3510 in rviz::RenderSystem::loadOgrePlugins() () from /opt/ros/melodic/lib/librviz.so
#8  0x00007ffff7ad7aa7 in rviz::RenderSystem::RenderSystem() () from /opt/ros/melodic/lib/librviz.so
#9  0x00007ffff7ad7c75 in rviz::RenderSystem::get() () from /opt/ros/melodic/lib/librviz.so
#10 0x00007ffff7ad2c11 in rviz::QtOgreRenderWindow::QtOgreRenderWindow(QWidget*) () from /opt/ros/melodic/lib/librviz.so
#11 0x00007ffff7af9ce9 in rviz::RenderPanel::RenderPanel(QWidget*) () from /opt/ros/melodic/lib/librviz.so
#12 0x00007ffff7b45d48 in rviz::VisualizationFrame::initialize(QString const&) () from /opt/ros/melodic/lib/librviz.so
#13 0x00007ffff7b501ce in rviz::VisualizerApp::init(int, char**) () from /opt/ros/melodic/lib/librviz.so
#14 0x0000555555554e86 in main (argc=2, argv=0x7fffffffe298) at /home/irt/rviz/src/src/rviz/main.cpp:40
```

```bash
dummy@dummy-DUMMY:~/rviz$ cd build_debug/ && gdb --args ./devel/lib/rviz/rviz -l
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./devel/lib/rviz/rviz...done.
(gdb) run
Starting program: /home/dummy/rviz/build_debug/devel/lib/rviz/rviz -l
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdc4c5700 (LWP 12991)]
[ INFO] [1542668463.576895821]: rviz version 1.13.1
[ INFO] [1542668463.576957496]: compiled against Qt version 5.9.5
[ INFO] [1542668463.576975196]: compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffda872700 (LWP 12999)]
[New Thread 0x7fffda071700 (LWP 13000)]
[New Thread 0x7fffd9870700 (LWP 13001)]
[New Thread 0x7fffd906f700 (LWP 13002)]
[New Thread 0x7fffcbfff700 (LWP 13003)]
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 0x7fffbc102700 (LWP 13010)]
[New Thread 0x7fffbb901700 (LWP 13011)]
[New Thread 0x7fffbb100700 (LWP 13012)]
[New Thread 0x7fffba8ff700 (LWP 13013)]
[New Thread 0x7fffba0fe700 (LWP 13014)]
[New Thread 0x7fffb98fd700 (LWP 13015)]
[New Thread 0x7fffb90fc700 (LWP 13016)]
[New Thread 0x7fffb88fb700 (LWP 13017)]
QXcbConnection: XCB error: 145 (Unknown), sequence: 171, resource id: 0, major code: 139 (Unknown), minor code: 20
[New Thread 0x7fff9bcff700 (LWP 13018)]
[New Thread 0x7fff9b4fe700 (LWP 13019)]
[New Thread 0x7fff9acfd700 (LWP 13020)]
[New Thread 0x7fff9a4fc700 (LWP 13021)]
[New Thread 0x7fff99cfb700 (LWP 13022)]
[New Thread 0x7fff994fa700 (LWP 13023)]
[New Thread 0x7fff98cf9700 (LWP 13024)]
[... 38 lines cut ...]
(gdb) bt
#0  0x00007fff7f233496 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#1  0x00007fff7f237ee0 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#2  0x00007fff7f25c933 in Ogre::GLRenderSystem::GLRenderSystem() () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#3  0x00007fff7f2576e4 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#4  0x00007ffff31ada26 in Ogre::Root::installPlugin(Ogre::Plugin*) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#5  0x00007fff7f24079a in dllStartPlugin () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#6  0x00007ffff31adc61 in Ogre::Root::loadPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff7ad3510 in rviz::RenderSystem::loadOgrePlugins() () from /opt/ros/melodic/lib/librviz.so
#8  0x00007ffff7ad7aa7 in rviz::RenderSystem::RenderSystem() () from /opt/ros/melodic/lib/librviz.so
#9  0x00007ffff7ad7c75 in rviz::RenderSystem::get() () from /opt/ros/melodic/lib/librviz.so
#10 0x00007ffff7ad2c11 in rviz::QtOgreRenderWindow::QtOgreRenderWindow(QWidget*) () from /opt/ros/melodic/lib/librviz.so
[... cut ...]
#12 0x00007ffff7b45d48 in rviz::VisualizationFrame::initialize(QString const&) () from /opt/ros/melodic/lib/librviz.so
#13 0x00007ffff7b501ce in rviz::VisualizerApp::init(int, char**) () from /opt/ros/melodic/lib/librviz.so
#14 0x0000555555554e86 in main (argc=2, argv=0x7fffffffe298) at /home/dummy/rviz/src/src/rviz/main.cpp:40
(gdb) q
A debugging session is active.

        Inferior 1 [process 12975] will be killed.

Quit anyway? (y or n) y
```

Does anyone know a fix? Next step is building Ogre from source I guess.

diogoalmeida, 2018-11-19:
My solution around this issue was making sure my laptop screen remained on even if docked. Every time I try launching RViz on the laptop without the main screen on, I'll get the segfault.

VictorLamoine, 2018-11-19:
The problem was the remote connection with MobaXterm. Using a screen solved it.

amandayeyan, 2021-02-01:
> The problem was the remote connection with MobaXterm

Hi Victor, I encounter the same problem and most probably is the mobxterm remote connection causing such issue. So what do you mean by using a screen?

VictorLamoine, 2021-02-01:
A physical screen 😄