ros-visualization/rviz #1160: Crashing when visualizing odom message with large covariance values

pl-kabaradjian, 2017-10-05:
Hello,

I am running kinetic and I am encountering crashes when trying to visualize odometry from kobuki/turtlebot. 
Here is the crash message:

> rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.

It seems to be caused by  kobuki sending large covariance values like "1.7976931348623157e+308".
Here is an example of kubuki odom message:

[image] 

I tried sending the same message but replacing these large values by 1 and Rviz doesn't crash anymore.

I also found a commit concerning a similar problem in rtbamap :
[link to the commit](https://github.com/introlab/rtabmap_ros/commit/8eea06c32beeb254a540c0f4afc11b87291afe8b)

It was working before so there might be an update or a change in turtlebot driver that broke something . I would really appreciate if someone could look into this.

Thanks in advance,

PLK

NiftyGrimoire, 2018-03-22:
I am also encountering this issue, with odom covariances reaching 1e+308.
Currently working around this issue by piping the odom (from the kobuki node) into another node, which limits the covariances.

rhaschke, 2019-09-24:
Fixed via #1400, released into 1.13.4.

closed 2019-09-24 by rhaschke

positron96, 2019-10-01:
Thanks! So, what version of ROS do I need to have a fixed rviz?

rhaschke, 2019-10-01:
Melodic. rviz 1.13.4 was synced on Monday, Sep. 30th.