ros-visualization/rviz #1169: Is There such a Thing as Visibility Box?

sradmard, 2017-11-02:
Hi,
This is more of a question than an issue. I rely on rviz as my only visualisation tool, and my robot moves in a very cluttered environment, which makes it hard to focus on the robot in rviz. I would like to define a visibility box around my robot base_frame that any items outside of this box will be invisible (transparent), and only the items inside this box will be visible. 
I have already set my Views type into ThirdPersonFollower, and have set its Target Frame as base_frame, but I still have a lot of clutters on the Z axis that occludes my robot. Being able to define a visibility box would allow me to narrow visible field in the Z direction, and be able to focus on the robot. 
I would appreciate any hints on potential solutions, or whether even rviz provides such capability or not.

sradmard, 2017-11-02:
I think I partially found the answer. I guess adjusting Near Clip Distance under camera view would do what I am looking for. However, is there a Far Clip Distance parameter that I can adjust as well? I cannot find that under the camera view.

dhood, 2017-12-20:
not currently, but it is supported by Ogre Camera objects so it just needs to be exposed. It's unlikely that we'll have time to make the changes ourselves but a pull request would be welcome if you are interested. You can copy [this commit that originally added near clip support](https://github.com/ros-visualization/rviz/commit/19030770ae3bb4960c743a237835597ab9759891) to see what files need to be touched

dhood, 2018-04-07:
I'll close this for now because it seems your concern was mostly resolved with the near clip distance.

closed 2018-04-07 by dhood