ros-visualization/rviz #1170: No matching fbConfigs or visuals found

byteofprash, 2017-11-15:
Hey Guys, 

I'm trying to run RViz inside a docker, and I'm getting this error. 
`root@aaedfc06138a:/# rosrun rviz rviz
`[ INFO] [1510757230.062006300]: rviz version 1.12.13`
`[ INFO] [1510757230.062071100]: compiled against Qt version 5.5.1`
`[ INFO] [1510757230.062092800]: compiled against OGRE version 1.9.0 (Ghadamon)`
`libGL error: No matching fbConfigs or visuals found`
`libGL error: failed to load driver: swrast`
`libGL error: No matching fbConfigs or visuals found`
`libGL error: failed to load driver: swrast`
`libGL error: No matching fbConfigs or visuals found`
`libGL error: failed to load driver: swrast`

Any help? 

Thanks.

plnegre, 2017-12-07:
Same issue here. Just following the official ros nvidia instructions from [here](http://wiki.ros.org/docker/Tutorials/Hardware%20Acceleration) gives me the same error.

dhood, 2018-02-15:
@byteofprash you mentioned you're on a mac, I don't have any specific suggestions for that.

@plnegre if you haven't already, you can try using `nvidia-docker` and installing `mesa-utils` in the container. https://github.com/osrf/car_demo/blob/203bcdfdbf7967c30ebf687b2fa69c51abe9f461/Dockerfile#L12 has an example of an image that runs gazebo and rviz.

Even though this isn't necessarily resolved I'm going to close this issue out because as far as I can tell the issue is not with rviz itself but rather the docker setup. You might be able to get better advice from other users on answers.ros.org.

closed 2018-02-15 by dhood

Baschdl, 2018-12-08:
@byteofprash Did you find a solution for that?

sunsided, 2020-05-21:
Probably outdated, but this problem still exists today. I eventually steered away from using the officially provided ROS (desktop full) images and built one myself.

In essence, rather than doing

```dockerfile
FROM osrf/ros:kinetic-desktop-full-xenial
```

I did

```dockerfile
FROM nvidia/cudagl:9.0-base-ubuntu16.04

ENV UBUNTU_RELEASE=xenial
RUN sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $UBUNTU_RELEASE main" > /etc/apt/sources.list.d/ros-latest.list'
RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654

RUN apt-get update && apt-get install -y \
    ros-kinetic-desktop-full \
 && rm -rf /var/lib/apt/lists/*
```

The remaining problems of passing X11 around still apply, but that is documented in the [Hardware Acceleration](http://wiki.ros.org/docker/Tutorials/Hardware%20Acceleration) tutorial. There is a bug though where it states `echo $xauth_list | xauth -f $XAUTH nmerge -` for Ubuntu, but requires `echo "$xauth_list"` - note the quotes.