ros-visualization/rviz #1171: Segmentation fault with VNC

plguhur, 2017-11-20:
rviz is working fine but I would like to use it in headless mode.
So I configured vnc4server.

When I launch vnc, I have this error: 

> ~$ rviz --verbose
> QXcbConnection: Failed to initialize XRandr
> Qt: XKEYBOARD extension not present on the X server.
> [ INFO] [1511197994.229080382]: rviz version 1.12.13
> [ INFO] [1511197994.229162184]: compiled against Qt version 5.5.1
> [ INFO] [1511197994.229193910]: compiled against OGRE version 1.9.0 (Ghadamon)
> Segmentation fault (core dumped)

Using gdb, I have this message:

> Starting program: /opt/ros/kinetic/bin/rviz 
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> [New Thread 0x7fffdd600700 (LWP 1315)]
> QXcbConnection: Failed to initialize XRandr
> Qt: XKEYBOARD extension not present on the X server.
> [ INFO] [1511197778.554813126]: rviz version 1.12.13
> [ INFO] [1511197778.554892702]: compiled against Qt version 5.5.1
> [ INFO] [1511197778.554921888]: compiled against OGRE version 1.9.0 (Ghadamon)
> [New Thread 0x7fffd5051700 (LWP 1445)]
> [New Thread 0x7fffd4850700 (LWP 1448)]
> [New Thread 0x7fffcffff700 (LWP 1449)]
> [New Thread 0x7fffcf7fe700 (LWP 1464)]
> 
> Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
> 0x0000000000000000 in ?? ()
> (gdb) bt
> #0  0x0000000000000000 in ?? ()
> #1  0x00007fffcd9aa6eb in ?? ()
>    from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
> #2  0x00007fffcd9b46e8 in ?? ()
>    from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
> #3  0x00007fffcd9a53dc in ?? ()
>    from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
> #4  0x00007fffcd9c3ae2 in Ogre::GLRenderSystem::_createRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*)
>     () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
> #5  0x00007ffff3d3588c in Ogre::Root::createRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*) ()
>    from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
> #6  0x00007ffff7ad74da in rviz::RenderSystem::tryMakeRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*, int) ()
>    from /opt/ros/kinetic/lib/librviz.so
> #7  0x00007ffff7ad8c9e in rviz::RenderSystem::makeRenderWindow(unsigned long, unsigned int, unsigned int, double) () from /opt/ros/kinetic/lib/librviz.so
> #8  0x00007ffff7adb175 in rviz::RenderSystem::RenderSystem() ()
>    from /opt/ros/kinetic/lib/librviz.so
> #9  0x00007ffff7adb2d5 in rviz::RenderSystem::get() ()
>    from /opt/ros/kinetic/lib/librviz.so
> #10 0x00007ffff7ad6691 in rviz::QtOgreRenderWindow::QtOgreRenderWindow(QWidget*) () from /opt/ros/kinetic/lib/librviz.so
> #11 0x00007ffff7afcd19 in rviz::RenderPanel::RenderPanel(QWidget*) ()
>    from /opt/ros/kinetic/lib/librviz.so
> #12 0x00007ffff7b47bc0 in rviz::VisualizationFrame::initialize(QString const&)
> ---Type <return> to continue, or q <return> to quit---
>     () from /opt/ros/kinetic/lib/librviz.so
> #13 0x00007ffff7b52e29 in rviz::VisualizerApp::init(int, char**) ()
>    from /opt/ros/kinetic/lib/librviz.so
> #14 0x0000000000400c90 in main ()

wtmilk, 2018-01-27:
have you fixed this problem?

plguhur, 2018-03-10:
No I stop using rviz on that remote computer

mei123hao, 2020-03-22:
same question. ASk for help!

rhaschke, 2020-03-23:
Usually, these issues are related to remote OpenGL usage. Doing heavy graphics output, as rviz does, isn't recommended over network anyway.