ros-visualization/rviz #1172: `load` member is only called once when creating/deleting a panel multiple times

VictorLamoine, 2017-11-21:
Minimal example package:
https://gitlab.com/InstitutMaupertuis/rviz_plugin_load_bug

I expect the `load` member of my RViz panels to be called each time the panel is created, it is not the case.

How to reproduce:
- Start RViz
- Add a custom panel with a `load(const rviz::Config& config)` member
- Delete the panel
- Add the same panel again

The `load` member will not be called after the first creation, resulting in a partially initialized panel.
This is problematic for me because I use the `load` member to make sure service connections are done etc...

I'm not sure if it is a bug or an intended behaviour.

wjwwood, 2018-05-10:
I'd say it's not a bug, but that this part of the system was just not designed to allow load to be called each time. However, I don't see why that must be the case. So I'll conservatively mark this as an enhancement, though I think there's a case for it being considered a bug as well. Either changing this would be possible if anyone wanted to pursue a pull request to that end.

VictorLamoine, 2018-10-19:
Closing because of the discussion in https://github.com/ros-visualization/rviz/pull/1300
This is not a bug but an intended behavior that makes sense.

closed 2018-10-19 by VictorLamoine