ros-visualization/rviz #1173: sometimes rviz fails to connect to /tf topic?

randaz81, 2017-11-29:
Sometimes I experience the following strange issue:
I launch rviz and and a simple tf publisher such as:
`rosrun tf static_transform_publisher 1 2 3 0 0 0 /world /robot 10`

now, if I do:
`rostopic echo /tf`
the transform is displayed correctly.

However, rviz seems not able to display the /tf, neither in the 3d view, nor in the tree visualization. Indeed roswtf displays an error message saying that the publisher node and rviz are not connected.

I cannot find a logic to explaining this weird behavior. Sometimes restarting rviz fixes the issue. Sometimes I need to perform a rosnode cleanup. Sometimes I have even to restart roscore.  
I wonder if someone else is experiencing the same bug.

I'm using ROS lunar with:
ros_comm 1.13.2-0xenial-20171012-115727-0800

Yeshasvitvs, 2019-04-18:
@randaz81 Did you manage to fix the problem ??

darshb34, 2020-08-19:
@Yeshasvitvs @randaz81 any hack around this?

Thanks