ros-visualization/rviz #1175: replace RenderPanel::sendMouseMoveEvent by using QWidget::setMouseTracking

simonschmeisser, 2017-12-04:
https://github.com/ros-visualization/rviz/blob/fa59398651327241e9c6d72019e5fa7731ea0b6e/src/rviz/render_panel.cpp#L91

uses a timer (33ms!) to fake receiving mouse move events when there is no button pressed (for interactive markers for example). Is there a reason why we are not setting QWidget::setMouseTracking
http://doc.qt.io/qt-5/qwidget.html#mouseTracking-prop ?

I'll open a PR later if no one remembers why it is done this way

dhood, 2017-12-20:
At first I wondered it if was a new addition to qt but it looks like it was available in at the time [of the commit](https://github.com/ros-visualization/rviz/commit/140d43b76d29469e33ac6c4819bb6caf1bc370c5). This approach might be appropriate if we were trying to limit the callback frequency or something, but at 30Hz it doesn't look like the case 😄. So it seems safe to me to swap to using `setMouseTracking(true)` (unless @hersh happens to remember a reason why it wasn't appropriate, but we're going back quite some years now).

Perhaps the reason will reveal itself if/when you give it a shot @simonschmeisser 😉

rhaschke, 2019-09-23:
@simonschmeisser, could you provide a PR for this please?

closed 2020-02-24 by simonschmeisser