ros-visualization/rviz #1176: when i use moveit for motion planning of pr2 robot than blank screen appears in rviz and in motion planning tab it shows following error: MotionPlanning The class required for this display, 'moveit_rviz_plugin/MotionPlanning', could not be loaded. Error: Failed to load library /opt/ros/kinetic/lib/libmoveit_motion_planning_rviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgproc3.so.3.3: cannot open shared object file: No such file or directory)

aqib005, 2017-12-18:


aqib005, 2017-12-18:
when i use moveit for motion planning of pr2 robot than blank screen appears in rviz and in motion planning tab it shows following error: MotionPlanning
The class required for this display, 'moveit_rviz_plugin/MotionPlanning', could not be loaded.
Error:
Failed to load library /opt/ros/kinetic/lib/libmoveit_motion_planning_rviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgproc3.so.3.3: cannot open shared object file: No such file or directory)

wjwwood, 2017-12-18:
Please do not post your entire problem in the title of the GitHub issue.

It looks like you do not have opencv installed. But since this has to do with moveit, you might want to ask on their lists or answers.ros.org first, see: http://wiki.ros.org/Support

closed 2017-12-18 by wjwwood