ros-visualization/rviz #1178: this commit cause some interactive marker not working

zhuyiif, 2017-12-29:
https://github.com/ros-visualization/rviz/commit/73ad3e122283b2a8cbd116bda0b2ccf63c3b03e7,
after I update rviz ,some interactive markers not working, even official tutorial e.g https://github.com/ros-visualization/visualization_tutorials.

rhaschke, 2017-12-29:
Indeed this breaks interactive markers of MoveIt! as well. Previously, rviz was sloppy about unnormalized unit quaternions. Now it rejects those markers. It would be nice to have a smooth transition phase, where a warning is generated, the quaternion is normalized and processed anyway.

rhaschke, 2017-12-29:
Actually, the offending PR is #1167.

zhuyiif, 2017-12-29:
thanks for quick response :)

rhaschke, 2018-01-09:
The problem is fixed and a release candidate is available via [ros-shadow-fixed](http://wiki.ros.org/ShadowRepository).

wjwwood, 2018-05-10:
Closing this assuming the issue was actually fixed by now. Please comment if that's not the case.

closed 2018-05-10 by wjwwood