ros-visualization/rviz #1181: Arrow Marker broken when using a start and end point instead of pose.

Levi-Armstrong, 2018-01-03:
It looks like with the latest change/addition of checking if the quaternion is valid has broken the arrow marker when using a start and end point. In the case of using a start and end point the pose has all zeros for the orientation which causes it to fail with a display error below. Now if I just set the pose orientation of my marker to (0,0,0,1) the marker shows but not correctly because it now uses the pose data instead of points. In the interim I have to calculate a pose for everything work.

```
Contains invalid quaternions (length not equal to 1)!
```

wjwwood, 2018-01-03:
@Levi-Armstrong We're aware of the break in behavior, we're trying to address it now:

https://github.com/ros-visualization/rviz/pull/1179

wjwwood, 2018-01-03:
We should test this case though as well, @Levi-Armstrong can you provide a bag file and/or an rviz config that reproduces it? That would save us some time in testing.

dhood, 2018-01-04:
If the message had an uninitialised pose before it would have been converted internally to (0, 0, 0, 1) so I would expect it to behave the same if you set an explicit orientation of (0, 0, 0, 1) (that's what I see with some manual testing of mine). I might be testing something other than what you've described, so I'd be interested in seeing the contents of the message to get a better idea of what's going on if you can share @Levi-Armstrong

keke8181, 2018-01-05:
@wjwwood I have the same problem. Rviz InteractiveMarker Status error: Marker contains invalid quaternions (length not equal to 1)! My rviz version is 1.12.14. But the same interactive marker works in the rviz version 1.12.13. Do you know how to install older version?

wjwwood, 2018-01-05:
@keke8181 we're already aware of the issue and we're working on a fix. You must build rviz from source if you want the older version.

blank970, 2018-01-08:
Did the problem fix?

dhood, 2018-01-08:
A fix has been released into the testing repos, see https://discourse.ros.org/t/fix-released-rviz-1-12-14-not-displaying-markers-etc-with-invalid-quaternions/3610

If you update publishers of invalid quaternions to publish valid (normalised) ones, this will restore the behaviour of rviz 1.12.13. You should do this regardless because rviz will reject invalid quaternions in the future. If that's not feasible you can get rviz 1.12.15 from the testing repo http://wiki.ros.org/ShadowRepository before it gets synced to the main apt repo.

keke8181, 2018-01-08:
@blank970 not yet

blank970, 2018-01-08:
how could i install the old version of rviz?

Levi-Armstrong, 2018-01-09:
@wjwwood Do you still need a bag file if so I should be able to get one generated today?

wjwwood, 2018-01-09:
@blank970 as I said above, to get an older version of rviz, you need to build it from source. As @dhood said, you may also switch to our testing repositories to get the new version which is not yet in public.

----

@Levi-Armstrong if you have a small bag, it would help us. We could possibly include it in some acceptance test or something like that. It would also just be find if you could test our fix for the issue to see if it also fixed this problem.

dhood, 2018-04-17:
closing this under the assumption it was fixed in 1.12.15

closed 2018-04-17 by dhood